carla-simulator/carla #8544: Do we have .whl file for carla-0.10-cp38-linux?

timmao955, 2025-01-09:
CARLA version: 0.10.0
Platform/OS: Ubuntu 20.04 release version

Hi, thanks for the update of carla 0.10.0!

I want to run carla on Ubuntu22.04 and use python3.8. From the release of carla 0.10.0, I find that windows version has the .whl file 'carla-0.10-cp38-cp38-win_amd64.whl', but linux version only have 'carla-0.10.0-cp310-cp310-linux_x86_64.whl'.

Do you plan to update the cp38 .whl file for linux? Thank you.

Blyron, 2025-01-09:
That's our bad, We will reupload the package soon, with all wheels included.

IrfanAli91170, 2025-01-10:
Excuse me. Where can you find this 'carla-0.10.0-cp310-cp310-linux_x86_64.whl' file, please ? 

I need this one on Linux version.

timmao955, 2025-01-10:
> Excuse me. Where can you find this 'carla-0.10.0-cp310-cp310-linux_x86_64.whl' file, please ? 
> 
> 
> 
> I need this one on Linux version.

In Carla-0.10.0-Linux-Shipping/PythonAPI/carla/dist