carla-simulator/carla #8554: Fatal error and crash when trying to spawn a camera

Vincent318, 2025-01-09:
Hello, I built Carla from source on Windows 10 and unfortunately, when I try to spawn a camera, the simulator crashes and there comes "Fatal error!". The code which produces this error is: 

import carla
import time

# Verbindung zum Carla-Server herstellen
client = carla.Client('localhost', 2000)
client.set_timeout(10.0)

# Welt und Blueprint-Bibliothek abrufen
world = client.get_world()
blueprint_library = world.get_blueprint_library()

# Kamera-Blueprint suchen
camera_bp = blueprint_library.find('sensor.camera.depth')

# Spawn-Position und Rotation der Kamera festlegen
camera_transform = carla.Transform(carla.Location(x=1.5, y=0, z=2.4))

# Kamera spawnen
camera = world.spawn_actor(camera_bp, camera_transform)

The last line camera = world.spawn_actor(camera_bp, camera_transform) is responsible for the error and the crash. When I leave out this line, there's neither a crash, nor a fatal error. In the log, there stands:

[2025.01.09-13.28.08:392][936]LogCarla: Sensor 1 '' created in primary server
[2025.01.09-13.28.08:442][942]LogCarla: Spawning actor 'vehicle.tesla.model3'
[2025.01.09-13.28.08:454][943]LogCarla: Spawning actor 'sensor.camera.rgb'
[2025.01.09-13.28.08:455][943]Cmd: g.TimeoutForBlockOnRenderFence 300000
[2025.01.09-13.28.08:455][943]g.TimeoutForBlockOnRenderFence = "300000"
[2025.01.09-13.28.18:736][943]LogWindows: Could not start crash report client using ../../../Engine/Binaries/Win64/CrashReportClientEditor-Win64-Debug.exe
[2025.01.09-13.28.18:736][943]LogMemory: Platform Memory Stats for Windows
[2025.01.09-13.28.18:736][943]LogMemory: Process Physical Memory: 13134.45 MB used, 13136.42 MB peak
[2025.01.09-13.28.18:736][943]LogMemory: Process Virtual Memory: 15820.44 MB used, 15822.41 MB peak
[2025.01.09-13.28.18:736][943]LogMemory: Physical Memory: 28154.75 MB used,  37116.84 MB free, 65271.59 MB total
[2025.01.09-13.28.18:736][943]LogMemory: Virtual Memory: 134174568.00 MB used,  43159.83 MB free, 134217728.00 MB total
[2025.01.09-13.28.18:736][943]Message dialog closed, result: Ok, title: The UE4-CarlaUE4 Editor has crashed and will close, text: Fatal error!

[2025.01.09-13.28.18:736][943]LogWindows: Error: === Critical error: ===
[2025.01.09-13.28.18:736][943]LogWindows: Error: 
[2025.01.09-13.28.18:736][943]LogWindows: Error: Fatal error!
[2025.01.09-13.28.18:736][943]LogWindows: Error: 
[2025.01.09-13.28.18:736][943]LogWindows: Error: Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address 0x00000000000002f4
[2025.01.09-13.28.18:736][943]LogWindows: Error: 
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3eaf3b7 UE4Editor-Carla.dll!AWeather::CheckWeatherPostProcessEffects() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Weather\Weather.cpp:29]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3eb7dde UE4Editor-Carla.dll!AWeather::NotifyWeather() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Weather\Weather.cpp:81]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3c990e9 UE4Editor-Carla.dll!ASceneCaptureSensor::BeginPlay() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Sensor\SceneCaptureSensor.cpp:587]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007ff81e322ccc UE4Editor-Engine.dll!AActor::DispatchBeginPlay() [C:\UnrealEngine\Engine\Source\Runtime\Engine\Private\Actor.cpp:3517]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007ff81e346957 UE4Editor-Engine.dll!AActor::PostActorConstruction() [C:\UnrealEngine\Engine\Source\Runtime\Engine\Private\Actor.cpp:3347]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007ff81e3306bc UE4Editor-Engine.dll!AActor::FinishSpawning() [C:\UnrealEngine\Engine\Source\Runtime\Engine\Private\Actor.cpp:3246]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007ff81ea6d4db UE4Editor-Engine.dll!UGameplayStatics::FinishSpawningActor() [C:\UnrealEngine\Engine\Source\Runtime\Engine\Private\GameplayStatics.cpp:693]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3cd2610 UE4Editor-Carla.dll!ASensorFactory::SpawnActor() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Sensor\SensorFactory.cpp:160]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3bef558 UE4Editor-Carla.dll!UE4Function_Private::TFunctionRefCaller<<lambda_e596a5a0e0e47125cd403ccb08d0926f>,FActorSpawnResult __cdecl(FTransform const &,FActorDescription const &)>::Call() [C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Templates\Function.h:539]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3c4faec UE4Editor-Carla.dll!UActorDispatcher::SpawnActor() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Actor\ActorDispatcher.cpp:62]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3c504b1 UE4Editor-Carla.dll!UCarlaEpisode::SpawnActorWithInfo() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Game\CarlaEpisode.cpp:449]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3dd478d UE4Editor-Carla.dll!<lambda_a19643c0d27391d6ddab825db58113c5>::operator()() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Server\CarlaServer.cpp:747]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3e7729e UE4Editor-Carla.dll!std::_Func_impl_no_alloc<<lambda_ae81b8460697fb21e5878a10e678ec30>,carla::rpc::Response<carla::rpc::Actor> >::_Do_call() [C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\INCLUDE\functional:823]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3e621fa UE4Editor-Carla.dll!std::_Packaged_state<carla::rpc::Response<carla::rpc::Actor> __cdecl(void)>::_Call_immediate() [C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\INCLUDE\future:485]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3e9346f UE4Editor-Carla.dll!boost::asio::detail::executor_op<boost::asio::detail::binder0<carla::detail::MoveWrapper<std::packaged_task<carla::rpc::Response<carla::rpc::Actor> __cdecl(void)> > >,std::allocator<void>,boost::asio::detail::win_iocp_operation>::do_complete() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\boost\asio\detail\executor_op.hpp:70]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3e978cf UE4Editor-Carla.dll!boost::asio::detail::win_iocp_io_context::do_one() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\boost\asio\detail\impl\win_iocp_io_context.ipp:473]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3d6dd14 UE4Editor-Carla.dll!boost::asio::io_context::run_until<std::chrono::steady_clock,std::chrono::duration<__int64,std::ratio<1,1000000000> > >() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\boost\asio\impl\io_context.hpp:73]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3e52f29 UE4Editor-Carla.dll!FCarlaServer::RunSome() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Server\CarlaServer.cpp:3140]
[2025.01.09-13.28.18:736][943]LogWindows: Error: [Callstack] 0x00007fffd3c2d1ee UE4Editor-Carla.dll!FCarlaEngine::OnPreTick() [C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Game\CarlaEngine.cpp:301]
[2025.01.09-13.28.18:737][943]LogWindows: Error: [Callstack] 0x00007fffd3bfddce UE4Editor-Carla.dll!TBaseRawMethodDelegateInstance<0,FCarlaEngine,void __cdecl(UWorld *,enum ELevelTick,float),FDefaultDelegateUserPolicy>::ExecuteIfSafe() [C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:469]
[2025.01.09-13.28.18:737][943]LogWindows: Error: [Callstack] 0x00007ff81ec1e0c6 UE4Editor-Engine.dll!TMulticastDelegate<void __cdecl(UWorld *,enum ELevelTick,float),FDefaultDelegateUserPolicy>::Broadcast() [C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateSignatureImpl.inl:955]
[2025.01.09-13.28.18:737][943]LogWindows: Error: [Callstack] 0x00007ff81ec637a7 UE4Editor-Engine.dll!UWorld::Tick() [C:\UnrealEngine\Engine\Source\Runtime\Engine\Private\LevelTick.cpp:1286]
[2025.01.09-13.28.18:737][943]LogWindows: Error: [Callstack] 0x00007ff81bd3a63b UE4Editor-UnrealEd.dll!UEditorEngine::Tick() [C:\UnrealEngine\Engine\Source\Editor\UnrealEd\Private\EditorEngine.cpp:1720]
[... cut ...]
[2025.01.09-13.28.18:737][943]LogWindows: Error: [Callstack] 0x00007ff885e5cc91 ntdll.dll!UnknownFunction []
[2025.01.09-13.28.18:737][943]LogWindows: Error: 
[2025.01.09-13.28.18:763][943]LogExit: Executing StaticShutdownAfterError
[2025.01.09-13.28.18:770][943]LogWindows: FPlatformMisc::RequestExit(1)
[2025.01.09-13.28.18:770][943]LogCore: Engine exit requested (reason: Win RequestExit)
[2025.01.09-13.28.18:782][943]Log file closed, 01/09/25 14:28:18

Does somebody know what could be a reason for this problem and how to solve this? Thank you for your answers and your help!

PabloVD, 2025-01-10:
According to the line `Error: Unhandled Exception: EXCEPTION_ACCESS_VIOLATION reading address `, this may be caused by the program trying to access a not valid memory address. [Here](https://forums.unrealengine.com/t/unhandled-exception-exception-access-violation-reading-address-0x0000000000000000/752651) there are several steps to deal with that problem,

Besides that, according to the log, you are spawning more actors, so the above script is not the only one running. Could you please provide us with more info regarding all the script you are running?

Also, to check if there is some problem related to your build, you could try running the script in a package version, see [here](https://github.com/carla-simulator/carla/releases). I was able to run without problems the script in carla 0.10.

What carla version are you using?

Vincent318, 2025-01-10:
Thank you for your answer. The log was from another code where also a car was involved, but when I leave this out and only execute the code I sent, exactly the same error and callstack occurs. I use Carla 09.15 and built it from source, the Client API Version and Simulator API Version both is 0.9.15-244-gb60034be6-dirty. When I download the package version Carla09.15 from the internet and execute it, I have Client API Version and Simulator API Version both is 0.9.15 without the ending and the code doesn't cause a mistake. Maybe, could it be that the build of Carla wasn't 100% right so that the simulator crashes when I spawn a camera?

PabloVD, 2025-01-14:
Yes, if the package works well then it is likely that there is some issue with your compilation. Make sure to compile the engine in a compatible branch with the carla 0.9.15 branch. 
If you are running from editor, you can also try to create your own package and run the same code in a packaged version with `make package` and test your code in a custom package.
Besides, given that the problem seems to be caused by trying to access a not valid memory address, try to follow the steps  from the link I posted in the previous comment.