carla-simulator/carla #8559: Incorrect semantic segmentation in mine map, 0.10.0

PabloVD, 2025-01-10:
In the mine map in 0.10.0, rocks and vegetation are not visible in the semantic segmentation. Moreover, the terrain color does not seem to be the proper one. May be due to the usage of semantic segmentation in landscapes.
[image]

yuanhaorannnnnn, 2025-01-16:
Yes, you'are right, models from landscape isn't compatible with segmentation camera.

utkuoguzman, 2025-02-01:
So what is the expected time to fix this? Or how can one overcome this? why "add rock/stone tag" is relevant to this?

yuanhaorannnnnn, 2025-02-05:
> So what is the expected time to fix this? Or how can one overcome this? why "add rock/stone tag" is relevant to this?

you can update the code and content and check the update

closed 2025-04-01 by Blyron