carla-simulator/carla #8561: ue5_window_build_fail

hzyzyhhj, 2025-01-12:
Launching Unreal Editor with CARLA...
[0/2] Re-checking globbed directories...
[0/5] Generating Includes.def, E:/CarlaUE5/Unreal/CarlaUnr...aUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Definitions.defCMake Error: failed to create symbolic link 'E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Includes.def': �ͻ���û���������Ȩ��

FAILED: Unreal/Includes.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Includes.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Includes.def Unreal/Libraries.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Libraries.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Libraries.def Unreal/Options.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Options.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Options.def Unreal/Definitions.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Definitions.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Definitions.def E:/CarlaUE5/Build/Unreal/Includes.def E:/CarlaUE5/Build/Unreal/Libraries.def E:/CarlaUE5/Build/Unreal/Options.def E:/CarlaUE5/Build/Unreal/Definitions.def
C:\Windows\system32\cmd.exe /C "cd /D E:\CarlaUE5\Build\Unreal && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Includes.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Includes.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Includes.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Includes.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Libraries.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Libraries.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Libraries.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Libraries.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Options.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Options.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Options.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Options.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Definitions.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/Carla/Definitions.def && "C:\Program Files\CMake\bin\cmake.exe" -E create_symlink E:/CarlaUE5/Build/Unreal/Definitions.def E:/CarlaUE5/Unreal/CarlaUnreal/Plugins/CarlaTools/Definitions.def"
ninja: build stopped: subcommand failed.

hzyzyhhj, 2025-01-12:
Has anyone encountered this error,？thank you

MarcelPiNacy-CVC, 2025-01-14:
Hi, you need to enable developer mode in the control panel. Otherwise CMake can't create symbolics links, which is currently used by the build.

closed 2025-01-14 by MarcelPiNacy-CVC