carla-simulator/carla #8576: PythonAPI in 0.10.0 docker

jnaor, 2025-01-15:


jnaor, 2025-01-15:
When I run `./CarlaUnreal.sh --ros2`, attach to the docker and run `python3 ros2_native.py ` I get the error:

```
Traceback (most recent call last):
  File "/home/carla/PythonAPI/examples/ros2/ros2_native.py", line 16, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'
```
I see the .whl file in `/home/carla/PythonAPI/carla/dist` but don't know how to install it. The docker image doesn't have pip, curl, wget or unzip.

Thanks!

timmao955, 2025-01-16:
Do you use the docker here? > https://carla-ue5.readthedocs.io/en/latest/build_docker/

If you put the docker on a server and run the client locally, you can change the `localhost` in your code. The default is `127.0.0.1` and change it to your server ip.

If you run both client and server in your docker, you should install some dependencies. Just like what you said, pip, curl, wget or unzip. Use `cd /home/carla/PythonAPI/carla/dist` and use `pip install carla-0.10-cp38-cp38-win_amd64.whl`. After that you can use `pip show carla` to check that whether 0.10.0 carla package is installed.

Hopes this helps.

jnaor, 2025-01-16:
Thanks @timmao955. I'm not exactly sure what you mean, I'm pulling the docker image using `docker pull`, not building from the Dockerfile. If I do that I guess I could add pip (or just install the wheel in the first place). 
My guess is that having the PythonAPI already working "our of the box" would be the desired behavior expected by most people pulling the docker image,