carla-simulator/carla #8577: ue5 collapse Assertion failed: ExternalPooledBuffer.IsValid()

bentleyer, 2025-01-16:
CARLA version:  ue5-dev
Platform/OS: windows11
Problem you have experienced:
when i compiled the ue5,and open a project the ue5 collapse ,and tell me 
Assertion failed: ExternalPooledBuffer.IsValid() [File:carla\UnrealEngine5_carla\Engine\Source\Runtime\RenderCore\Private\RenderGraphBuilder.cpp] [Line: 1096] 
Attempted to register NULL external buffer.

What you expected to happen:
 i can open the project correct
Steps to reproduce:
Building CARLA in Windows with Unreal Engine 5.5  according to the document: 
https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/

Other information (documentation you consulted, workarounds you tried):

[image]

bentleyer, 2025-01-16:
[image]

bentleyer, 2025-01-16:
i download all the assert and rebuilt the unreal , and it work! maybe i should clone the assert first

closed 2025-01-16 by bentleyer

JaneyJin, 2025-01-22:
Hello @bentleyer, can you elaborate what mean by download all assert? Thanks in advance.

bentleyer, 2025-01-24:
> Hello [@bentleyer](https://github.com/bentleyer), can you elaborate what mean by download all assert? Thanks in advance.

sorry, i spell error, i download all the assets according to the doc

[image]

artificialhumanbeing, 2025-02-07:
@bentleyer Hi, it's slow to download the assets.

bentleyer, 2025-02-10:
> [@bentleyer](https://github.com/bentleyer) Hi, it's slow to download the assets. Can you share the Carla-content of ue5-dev ? Please. Thank you.

sorry,，it is too large, and i do not konw how to share you maybe you could clone the repository 

https://bitbucket.org/carla-simulator/carla-content/src/ue5-dev/

cdcstar, 2025-02-26:
@artificialhumanbeing   do you have solved this problem

artificialhumanbeing, 2025-02-26:
> [@artificialhumanbeing](https://github.com/artificialhumanbeing) do you have solved this problem
Download  from https://bitbucket.org/carla-simulator/carla-content/get/d8db6b0e4b02261499327259719bb8a97ff4f602.zip. And enter this file. Then enter  `git lfs fetch` and `git lfs checkout` .

cdcstar, 2025-02-27:
@artificialhumanbeing  i have build and run scuessfully i want know,  your UE version is 5.5.0

martelzhang, 2025-03-02:
what is asset mean? You mean the carla-content? I clone all the content. But still have this crash. Appreciate to ur reply.

bentleyer, 2025-03-02:
> what is asset mean? You mean the carla-content? I clone all the content. But still have this crash. Appreciate to ur reply.

yeah，i download the content from https://bitbucket.org/carla-simulator/carla-content/src/ue5-dev, you carla branch is ue5-dev?