carla-simulator/carla #8582: ./CarlaSetup.sh: line 113: cmake: command not found

XLR8-VISION, 2025-01-16:


MarcelPiNacy-CVC, 2025-01-16:
Can you verify if UE5 is properly built? Try running `./Setup.sh && ./GenerateProjectFiles.sh && make` in the unreal engine folder (it seems to be `/home/ubuntu/CarlaUE5`). Maybe a previous setup failed and left UE5 in a partially built state. I will add an argument to CarlaSetup to handle this case later.

MarcelPiNacy-CVC, 2025-01-16:
You don't have the correct path in CARLA_UNREAL_ENGINE_PATH. Update the variable to point to where you UE5 resides. It might be in ../UnrealEngine5_carla relative to CARLA's directory.

XLR8-VISION, 2025-01-20:
**Hi @MarcelPiNacy-CVC** 

It is pretty confusing, because the wrong folder is named CarlaUE5 and the correct folder is named UnrealEngine.

We can close this issue now as solved. Thank you for your quick support!

closed 2025-01-20 by MarcelPiNacy-CVC