carla-simulator/carla #8584: CARLA 0.9.15 Crash - VulkanRHI vkCreateDescriptorPool Error

SBAccn, 2025-01-17:
I’m experiencing a crash when trying to launch CARLA 0.9.15 using the CarlaUE4.sh script even with the -RenderOffScreen option. The terminal outputs the following error message:

./CarlaUE4.sh -RenderOffScreen
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 803]
VulkanRHI::vkCreateDescriptorPool(Device->GetInstanceHandle(), &PoolInfo, VULKAN_CPU_ALLOCATOR, &DescriptorPool) failed, VkResult=-2
at /home/jenkins/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanPendingState.cpp:52
with error VK_ERROR_OUT_OF_DEVICE_MEMORY
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault

System Information
Machine is Dell XPS 13
OS : Ubuntu 22.04
CARLA Version: 0.9.15GPU id = 0 (Intel(R) Arc(tm) Graphics (MTL))
Device Name: Intel(R) Arc(tm) Graphics (MTL)
Vulkan Version: 1.3.204
Vulkan Driver Version: 1.3.204

[image]

PatrickPromitzer, 2025-01-20:
Hi,
On the Dell website I read that your Laptop (XPS 13-Laptop) should have an "qualcomm adreno" GPU.
I found one version of this GPU with some GPU RAM data, and it was <8GB.

Carla needs at least 6GB GPU RAM, and 8Gb would be better to prevent crashes.

My guess is that your Laptop doesn't have enough power to run Carla.

SBAccn, 2025-01-22:
> Hi, On the Dell website I read that your Laptop (XPS 13-Laptop) should have an "qualcomm adreno" GPU. I found one version of this GPU with some GPU RAM data, and it was <8GB.
>
> Carla needs at least [...]

Thanks for your efforts and time, I have been stuck here for the past 2 weeks , I really appreciate it @PatrickPromitzer 
Could you help me with this further query as well ?
But in my other machine , the same script seems to run for at least a couple of times before it crashes ,

Specifications of it are , 
Lenovo Thinkpad T14 Gen3
12th Gen i5 Processor
16 GB RAM
Vulkan Version: 1.3.204
Vulkan Driver Version: 1.3.204
OS : Ubuntu 22.04
CARLA Version: 0.9.15
GPU id : Intel(R)  Graphics (ADL GT2))

PatrickPromitzer, 2025-01-22:
You need to check if your laptop has an GPU or is using the CPU as graphic card.

SBAccn, 2025-01-22:
lspci | grep -i vga
I ran this command on my machine and the results were 

@1TX7V44:~$ lspci | grep -i vga
00:02.0 VGA compatible controller: Intel Corporation Device 7d55 (rev 08) on my DELL XPS 13
00:02.0 VGA compatible controller: Intel Corporation Device 46a8 (rev 0c) on my Lenovo ThinkPad T14 , i looked up and both of these seems to be integrated GPUs

PatrickPromitzer, 2025-01-22:
Your hardware will struggle to run Carla if it can even run it at all, with only an CPU graphic and without an NVIDIA GPU.
We tried a laptop NVIDIA GPU at work, and even than Carla will crash if the laptop is not powerful enough.

You need a good hardware or Carla will probably not even start.

closed 2025-04-10 by MarcelPiNacy-CVC