carla-simulator/carla #8586: Can anyone open other town normally for carla 0.10.0?

SHWplus, 2025-01-17:
I successfully built the CARLA for unreal engine 5.5 and try to open Town01 Level. However, something wrong happened as follows:

CreateExport: Failed to load Outer for resource 'ExponentialHeightFog': BP_Sky_C /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2

and the scene was totally white

Blyron, 2025-01-17:
This is expected as the post processing values we are using in our camera sensors are adjusted for the visuals of town10. Try by decreasing light components intensity

SHWplus, 2025-01-17:
The error seems to be caused by lacking of some blueprints(6 Load Errors when open the level)? The skylight,Directional Light and all the lighting component could not be modified.

The load Error messages are as follows:
CreateExport: Failed to load Outer for resource 'ExponentialHeightFog': BP_Sky_C /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2

CreateExport: Failed to load Outer for resource 'Planet': BP_Sky_C /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2

CreateExport: Failed to load Outer for resource 'MaterialInstanceDynamic_0': StaticMeshComponent /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2.Planet

CreateExport: Failed to load Outer for resource 'NoWorldOffsetScene': BP_Sky_C /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2

CreateExport: Failed to load Outer for resource 'PostProcess': BP_Sky_C /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2

CreateExport: Failed to load Outer for resource 'SkyAtmosphere': BP_Sky_C /Game/Carla/Maps/Sublevels/Town01_Opt/T01_Rendering_and_Lightning_components.T01_Rendering_and_Lightning_components:PersistentLevel.BP_Sky_2

> This is expected as the post processing values we are using in our camera sensors are adjusted for the visuals of town10. Try by decreasing light components intensity

Blyron, 2025-01-17:
This is because the weather system has been changed.

SHWplus, 2025-01-17:
> This is because the weather system has been changed.

How can I fix that? Do I need to download something missing?