carla-simulator/carla #8589: ValueError: role_name: colors must have 3 channels (R,G,B)

shahbazsiddique69, 2025-01-18:
hi, i have built carla ue4 editor for 0.9.15. But, whenever i try to run any examples, i get this error:
ValueError: role_name: colors must have 3 channels (R,G,B)
i tried to follow the solution in github carla issues #8035 (https://github.com/carla-simulator/carla/issues/8035#issuecomment-2320778135)

But, when i command "git fetch --all --tags" in a terminal, i get this in return: "fatal: not a git repository (or any of the parent directories): .git"

Previously, i got this fatal error during make PythonAPI command, but I solved that error by updating the setup.sh file and mentioning the carla version -> 
#CARLA_VERSION=$(get_git_repository_version) in line 878 of Setup.sh .
change it to your Carla version and it would be work.
I've changed it to CARLA_VERSION="0.9.15".)

Using on Ubuntu 20.04
carla 0.9.15 on ue4

PabloVD, 2025-01-23:
This may be caused due to different conflicting carla versions. Ensure that you do not have any different carla python-api version installed by running `pip uninstall carla`. Then, create the python wheels running `make PythonAPI` and launch the editor with `make launch`.

shahbazsiddique69, 2025-01-25:
@PabloVD 
thanks! it worked

closed 2025-01-25 by shahbazsiddique69