carla-simulator/carla #8590: How do I adjust the weather in carla 0.10.0

KJ-Chang, 2025-01-20:
In carla 0.9.15 it was possible to adjust the weather `dynamic_weather.py`, But in 0.10.0 this file could not be found.
I tried changing the weather with`world.set_weather()` but it didn't seem to work.

dcratliff19, 2025-01-20:
I'm also having this issue on windows 11 with 0.10.0

- I'm not able to find the weather component in the map components. 
- If I add it from the content browser (CarlaUE5\Unreal\CarlaUnreal\Content\Carla\Blueprints\Weather\BP_CarlaWeather.uasset) , the sun and fog cannot be adjusted. However, the precipitation works just fine.

Blyron, 2025-01-20:
Weather cannot be adjusted right now in 0.10.0

closed 2025-01-20 by Blyron

dcratliff19, 2025-01-20:
Thanks for the reply!

KJ-Chang, 2025-01-21:
Thanks for the reply.