carla-simulator/carla #8591: Having trouble with building CARLA UE5

goosebuums, 2025-01-20:
Hello, I am trying to build carla UE5 0.10.0 with my ubuntu 22.04 computer
my computer 
CPU:Intel® Core™ i9-14900KF × 32
GPU: Geforce RTX4060Ti
RAM:128GB

I have an trouble executing this command:./CarlaSetup.sh --interactive

and the full error popping in the terminal is like this:

WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.
+ CMAKE_MINIMUM_VERSION=3.28.0
+ check_cmake_version 3.28.0 cmake
++ cmake --version
++ grep -Eo '[0-9]+\.[0-9]+\.[0-9]+'
+ CMAKE_VERSION=3.31.4
+ CMAKE_MINIMUM_VERSION=3.28.0
+ MAJOR=3
+ REMAINDER=31.4
+ MINOR=31
+ REVISION=4
+ MINIMUM_MAJOR=3
+ MINIMUM_REMAINDER=28.0
+ MINIMUM_MINOR=28
+ '[' -z 3.31.4 ']'
+ '[' 3 -gt 3 ']'
+ '[' 3 -eq 3 ']'
+ '[' 31 -gt 28 ']'
+ true
+ echo 'Found CMake 3.28.0'
Found CMake 3.28.0
Found CARLA content.
Found CARLA Unreal Engine at /home/a/CarlaUE5/UnrealEngine5_carla. Assuming already built...
Configuring the CARLA CMake project...
CMake Error at CMake/LinuxToolchain.cmake:23 (message):
  The specified Carla Unreal Engine 5 path does not exist ("").
Call Stack (most recent call first):
  Build/CMakeFiles/3.31.4/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:40 (project)

-- Configuring incomplete, errors occurred!

I think the bashrc is written properly, and I have no idea what is wrong with the system.

#CARLA CMake 3.28.3
PATH=/opt/cmake-3.28.3-linux-x86_64/bin:$PATH

#CARLA UnrealEngine5

#CARLA UnrealEngine5
export CARLA_UNREAL_ENGINE_PATH=/home/a/UnrealEngine5_carla

export CARLA_UNREAL_ENGINE_PATH=/home/a/CarlaUE5/UnrealEngine5_carla
export UE_ROOT=/home/a/SUnrealEngine5_carla

these are the bashrc codes for the environment settings. 

Hope I can solve this ....

PabloVD, 2025-01-21:
Note that you have two different `export CARLA_UNREAL_ENGINE_PATH`. Also, you need to build the UE5 engine before building Carla, have you done that? Ensure that you are in the `ue5-dev-carla` branch of the engine, using [this fork](https://github.com/CarlaUnreal/UnrealEngine).

MarcelPiNacy-CVC, 2025-02-03:
Hi, I'm closing this for now. Feel free to reopen it if this problem persists.

closed 2025-02-03 by MarcelPiNacy-CVC

JackYuan18, 2025-02-06:
Hello, I am trying to build carla 0.10.0 too. Was able to build pythonAPI but always failed at make launch. Any ideas?

sudo cmake --build Build --target launch
[0/2] Re-checking globbed directories...
[0/15] Performing prelaunch check for CARLA Content presence.
[2/15] Performing update step for 'fastdds'
-- Fetching latest from the remote origin
[4/15] Performing configure step for 'fastdds'
FAILED: Ros2Native/fastdds-prefix/src/fastdds-stamp/fastdds-configure /home/aCarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-stamp/fastdds-configure 
cd /home/a/CarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-build && /opt/cmake-3.30.7-linux-x86_64/bin/cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/a/CarlaUE5/Build/Ros2Native/install -DCMAKE_TOOLCHAIN_FILE=/home/zyuan/CarlaUE5/CMake/Toolchain.cmake -DBUILD_SHARED_LIBS=ON -DBUILD_TESTING=OFF -DCOMPILE_EXAMPLES=OFF -DCOMPILE_TOOLS=OFF -DTHIRDPARTY_Asio=FORCE -DTHIRDPARTY_fastcdr=FORCE -DTHIRDPARTY_TinyXML2=FORCE -DSQLITE3_SUPPORT=OFF -DOPENSSL_FOUND:BOOL=ON -DOPENSSL_INCLUDE_DIR:FILEPATH=/home/zyuan/UnrealEngine5_carla/Engine/Source/ThirdParty/OpenSSL/1.1.1t/include/Unix -DOPENSSL_SSL_LIBRARY:FILEPATH=/home/zyuan/UnrealEngine5_carla/Engine/Source/ThirdParty/OpenSSL/1.1.1t/lib/Unix/x86_64-unknown-linux-gnu/libssl.a -DOPENSSL_CRYPTO_LIBRARY:FILEPATH=/home/zyuan/UnrealEngine5_carla/Engine/Source/ThirdParty/OpenSSL/1.1.1t/lib/Unix/x86_64-unknown-linux-gnu/libcrypto.a -GNinja -S /home/zyuan/CarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds -B /home/zyuan/CarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-build && /opt/cmake-3.30.7-linux-x86_64/bin/cmake -E touch /home/zyuan/CarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-stamp/fastdds-configure
CMake Error at /home/a/CarlaUE5/CMake/Toolchain.cmake:25 (message):
  The specified Carla Unreal Engine 5 path does not exist ("").
Call Stack (most recent call first):
  /home/aCarlaUE5/Build/Ros2Native/fastdds-prefix/src/fastdds-build/CMakeFiles/3.30.7/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:33 (project)

-- Configuring incomplete, errors occurred!
ninja: build stopped: subcommand failed.

I think the bashrc is also correct as I ran ./CarlaSetup.sh before, which automatically installed UE5 for me. But after installation, carla is just not launched as described by the tutorial.

#CARLA CMake 3.28.3
PATH=/opt/cmake-3.28.3-linux-x86_64/bin:$PATH

export PYENV_ROOT="$HOME/.pyenv"
export PATH="$PYENV_ROOT/bin:$PATH"
eval "$(pyenv init --path)"
eval "$(pyenv init -)"

#CARLA UnrealEngine5
export CARLA_UNREAL_ENGINE_PATH=/home/zyuan/UnrealEngine5_carla

Any suggestion will be greatly appreciated!

sdwyc, 2025-02-10:
@JackYuan18 May be reopen terminal to work it, unless your $CARLA_UNREAL_ENGINE_PATH is correct.

Ftoy99, 2025-02-14:
sudo -E cmake --build Build --target launch you have set the path on the user but you are you using sudo without transfering the env variables

JayZ-97, 2025-03-04:
Windows 11
Visual Studio 2022（
Python 3.9
Unreal Engine 5.5
Git
CMake 3.22+
Ninja
Windows 11 SDK

Download the Asset source code from https://github.com/carla-simulator/carla/releases (direct download is a pre-compiled version, you can't use the UE editor).

Set the environment variable: CARLA_UNREAL_ENGINE_PATH “F:\EPIC\UE_5.5” on your system.
Enter the CARLA folder
Create a Python 3.9 virtual environment python -m venv venv39
Activate venv39\Scripts\activate.bat

Run
1. cmake -G Ninja -S . -B Build ^
  -DCMAKE_BUILD_TYPE=Release ^
  -DBUILD_CARLA_UNREAL=ON ^
  -DCARLA_UNREAL_ENGINE_PATH=%CARLA_UNREAL_ENGINE_PATH% ^
  -DCMAKE_Fortran_COMPILER= ^
  -DCMAKE_C_FLAGS="/utf-8" ^
  -DCMAKE_CXX_FLAGS="/utf-8" ^

2. cmake --build Build

Both steps work, but cmake --build Build --target launch ends up compiling 163/165 with an error that occurs magically. The error is: std::atomic_store_expression 
   std::atomic_store_explicit and std::atomic_load_explicit have been deprecated by C++20, so if I want to fix the error, I have to change them to store and load, but after changing the previous step cmake --build Build gives me a problem again, because it's using the old versions of store and load.

I think the compilation steps mentioned in the source code of carla0.10.0 are not well compatible with UE5.5 version