carla-simulator/carla #8594: Cannot launch Carla UE5 on Windows 11

kmsgm, 2025-01-21:
Hello, I am trying to launch Carla server on Windows 11 with Unreal Engine 5. But I keep getting the below log. How can I launch the Carla server with cmake? Or how can I launch the serve manually?

```bash
C:\Users\a\Carla\CarlaUE5>cmake --build Build --target launch
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?
```

MarcelPiNacy-CVC, 2025-01-27:
You need to run the cmake command in the root CARLA folder (which may be one level up from where you're currently running it by the log you shared). Feel free to reopen this issue if it still occurs.

closed 2025-01-27 by MarcelPiNacy-CVC

granddrumsco, 2025-01-29:
same problem

aaadam3042, 2025-04-30:
View my first comment on #8614 if problem still persists for others.