carla-simulator/carla #8599: CMake Error at CMake/Util.cmake:24 (message):   Missing CARLA content. Call Stack (most recent call first):   Unreal/CMakeLists.txt:53 (carla_error)

houaleframzi, 2025-01-22:
I'm trying to build Carla, and I've got the following error after cloning the Carla files into the content folder and executing "cmake --preset Linux-Release"

CMake Error at CMake/Util.cmake:24 (message):   Missing CARLA content. Call Stack (most recent call first):   Unreal/CMakeLists.txt:53 (carla_error)

note: I've downloaded all of Carla content using :" git clone --single-branch --depth 1 -b ue5-dev https://bitbucket.org/carla-simulator/carla-content.git Carla "

I'm using Ubuntu:22.4 Carla 0.10.0 with UE 5.5

closed 2025-01-22 by houaleframzi

hungle16, 2025-01-30:
did you find the solution to this problem?

houaleframzi, 2025-01-30:
yes, I had to add the UE5 path to the environment.

JayZ-97, 2025-02-27:
Can you please tell me how you solved this problem, the same problem happened when I compiled in windows with visual studio 2022.