carla-simulator/carla #8601: How to import additional maps in carla ue5 source code

timmao955, 2025-01-23:
Hi, I'm trying to import my maps into carla ue5 by using `make import`. However, the makefile says `MakeFile build is currently disabled.` Can we use other method to import additional maps? Thank you.

PatrickPromitzer, 2025-01-23:
Hi,
if I remember it correctly, Carla ue5 doesn't support map import currently.

You can copy the Carla ue4 maps into Carla ue5, but you have to make some changes to make them work (weather actor needs to be replaced, and other things).

Another idea would be to copy Town10 and modify it it. For example, delete every Building, ground, and other objects, and import new one in Unreal.

nigel-bruce-khan, 2025-02-14:
Hi, Im trying to copy a custom Map i made using Carla ue4 to Carla ue5, but it shows a black screen when i launch the map. Is it possible you could elaborate what needs to be done here "but you have to make some changes to make them work (weather actor needs to be replaced, and other things)." 

I copied the following files to the folders in Carla UE5, where I saw the same files for default maps: custom_map.uexp, custom_map.umap, custom_map.bin, custom_map.xodr, custom_map.Package.json

I am using the docker image of the latest Carla. I would really appreciate the help. Thanks!

PatrickPromitzer, 2025-02-18:
> Is it possible you could elaborate what needs to be done here "but you have to make some changes to make them work (weather actor needs to be replaced, and other things)."

Hi,
sorry but it isn't.
I have no idea which parts have to be replaced or added. The official documentation isn't done and I didn't work with Carl 0.10.0 to know.

You could wait until someone in the Carla team is seeing this issue, or join the Discord server and ask there.

Blyron, 2025-02-20:
> Hi, Im trying to copy a custom Map i made using Carla ue4 to Carla ue5, but it shows a black screen when i launch the map. Is it possible you could elaborate what needs to be done here "but you have [...]

You need to migrate the whole map. Before doing that I would suggest to clean all custom BPs or CARLA BPs from your UE4 map, then migrate the umap asset with Unreal integrated tool to migrate assets;
https://www.youtube.com/watch?v=UyKD_C4sFMc&ab_channel=NumenBrothers
Then copy the xodr.

nigel-bruce-khan, 2025-02-20:
Thanks for you reply. Just for clarification you are suggesting I go into the following folder in the map i have generated in Carla UE4 and delete its contents: "custom_maptar.gz\CarlaUE4\Content\Carla\Blueprints"

Then, since I am using a docker container of carla UE5, i should install this migration tool (or Unreal Engine 5 completely) and run it on the umap asset of my custom_map, and then copy the xodr file?

Blyron, 2025-02-20:
> Thanks for you reply. Just for clarification you are suggesting I go into the following folder in the map i have generated in Carla UE4 and delete its contents: [...]

You do not need to delete it. If you do not want to.  But yes I am suggesting you need to use UE migration tool and then copy the xodr manually because the xodr won't be copied properly

closed 2025-04-06 by Blyron

zmd-cc, 2025-04-21:
@nigel-bruce-khan
Hi, I encountered the same problem with a black screen. Could you please give me any hints about how to migrate?

nigel-bruce-khan, 2025-04-30:
Hi, I shifted to Unreal Engine 5.5 completly and then work on Carla10 through it. I can give you a list of steps but I don't remember if they are 100% correct so you might have to make some changes here and there:

First open unreal engine:

**/carla10/UnrealEngine5_carla/Engine/Binaries/Linux/UnrealEditor /carla10/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -vulkan -prefernvidia --ros2**

Building map in Carla:
1.	Use whats in this link but change as needed.

[Alternative methods to import maps - CARLA Simulator](https://carla.readthedocs.io/en/latest/tuto_M_add_map_alternative/#alternative-methods-to-import-maps)

2.	You double click an asset to open it in the view panel.
3.	In step 4 we import the fbx file to the new foder
4.	In step 5 we change what is possible to change in the FBX import options not all are possible, normal import method for example was possible but was configured to be different but I found it by hovering the mouse over the options to find out what they mean.
5.	For step 13, I made a new folder in Static giving it the name of the map example map_hall, and moved the static mesh there which was in the maps folder. If you get a notification when you do it, say yes.
6.	Step 14: Look for Open Drive Actor if you cant find it google it, I don’t remember how I found it. And then I couldn’t click generate routes, but there was an option for custom_path so I chose that.
7.	Then I went to this link (not sure if its needed): 

[Generate Pedestrian Navigation - CARLA Simulator](https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/)

In it I just followed step 3, selected everything in the Outliner on the right of the screen, and did Carla Exporter. I didn’t do the rest of the steps
8.	Then I pressed the green play button and in the terminal running unreal I saw what errors were generated. First was it said xodr missing. I figured out that I need to make an OpenDrive folder in Maps/map_folder and copy the map_hall.xodr file there as well. I did this by looking at the other map such as Town11_HD in its folder.

https://github.com/carla-simulator/carla/issues/4514

9.	Then I saw that there was an error about vehicles.json so in CarlaUE5/Unreal/CarlaUnreal/Content/Carla/Config/ I saw a Mine_01 folder containg Vehicles.json, so I added a folder map_hall similarly and copied the Vehicles.json file from Mine_01 folder to it. 
10.	Lastly for the weather class error I opened Town11 asset in the view panel and saw it had a weather folder with BP_Carla_Sky, so I opened map_hall in the view panel, and went to content browser at the bottom of the screen, then to Content/Carla/Blueprints/LevelDesign and dragged BP_Carla_Sky to the map viewing screen to add it. 
I changed the forwardLightingPriority of the BP_Carla_sky to 1 because there was some notification when I was pressing play. You can find it in the Outliner panel when you click details for BP_Carla_sky and search it there.
11.	I also saw the weather error when I pressed play on the Town11_hd map also so I guess it doesn’t matter
Then in the end I pressed the play button and it ran, so then I tried running manual_control.py in a separate window and it worked on the map_hall map

Note: you can drag things into the window showing the map to add them to the map, like lighting parameters, and the mesh you create.

zmd-cc, 2025-05-06:
@nigel-bruce-khan
That's very informative. Thank you so much and I will try step by step!

nallen94, 2025-08-15:
> Hi, sorry but it isn't. I have no idea which parts have to be replaced or added. The official documentation isn't done and I didn't work with Carl 0.10.0 to know.
>
> You could wait until someone in [...]

What is the link to the discord server ?