carla-simulator/carla #8604: lidar image acquisition faliure

little-win, 2025-01-23:
hello, when i use carla for simple simulation. i use the rgb and lidar dectors. but we i see the download lidar image. i notice that they are uncomplete. they are tooo small. and can anybody can help me? i want get some lidar which can be used.
# 
lidar_bp = blueprint_library.find('sensor.lidar.ray_cast')

# 
lidar_bp.set_attribute('range', '30')  # 
lidar_bp.set_attribute('channels', '32')  # 
lidar_bp.set_attribute('rotation_frequency', '10')  # 
lidar_bp.set_attribute('points_per_second', '1000000')  # 

# 
lidar_bp.set_attribute('horizontal_fov', '90')  

# 
lidar_bp.set_attribute('upper_fov', '10')
lidar_bp.set_attribute('lower_fov', '-10')

lidar_transform = carla.Transform(carla.Location(x=0, z=2.5))  # 
lidar_sensor = world.spawn_actor(lidar_bp, lidar_transform, attach_to=vehicles[0])

# 
def save_lidar_image(data):
    timestamp = data.timestamp
    filename = os.path.join(lidar_dir, 'lidar_%06d.ply' % timestamp)
    data.save_to_disk(filename)

[image]

[image]

little-win, 2025-01-24:
> Hi, you get this behavior because you are not using synchronous mode, and the rotation frequency is not synchronized with the tick rate of the simulation.
> 
> Here is a example for synchrounous_mode https://github.com/carla-simulator/carla/blob/ue5-dev/PythonAPI/examples/sensor_synchronization.py
> 
> Make sure that the sensor_tick in the Lidar settings, and fixed_delta_seconds for the word settings, are the same value

thank for your kindly reply， when i use the synchrounous_mode i found that the progress burn out for some times 

> `original_settings = world.get_settings()
    settings = world.get_settings()
    settings.synchronous_mode = True
    settings.fixed_delta_seconds = 0.5  # 可以调整为更小的值
    settings.max_substeps = 1  # 增加子步进的数量
    settings.max_substep_delta_time = 0.5  # 每个子步进的最大时间步长
    world.apply_settings(settings)
            #将同步模式设置为True
    #因为同步模式下车辆要使用自动模式必须开启同步模式的traffic manager。
    traffic_manager = client.get_trafficmanager()
    traffic_manager.set_synchronous_mode(True)

    blueprint_library = world.get_blueprint_library()

    try:
        # 获取车辆蓝图
        vehicle_bp1 = blueprint_library.filter('vehicle')[0]  # 第一种车类型
        vehicle_bp2 = blueprint_library.filter('vehicle')[1]  # 第二种车类型

        # 获取交通管理
        tm = client.get_trafficmanager()

        # 车辆生成位置
        vehicle_locations = [
            carla.Transform(carla.Location(x=394, y=289, z=0), carla.Rotation(yaw=90)),
            carla.Transform(carla.Location(x=394, y=281, z=0), carla.Rotation(yaw=90))
        ]

        vehicles = []

        # 生成第一辆车（Audi TT）
        vehicle1 = world.try_spawn_actor(vehicle_bp1, vehicle_locations[0])
        if vehicle1:
            vehicle1.set_autopilot(True)
            tm.distance_to_leading_vehicle(vehicle1, 0)
            vehicles.append(vehicle1)

        # 生成第二辆车（Tesla Model 3）
        vehicle2 = world.try_spawn_actor(vehicle_bp2, vehicle_locations[1])
        if vehicle2:
            vehicle2.set_autopilot(True)
            tm.distance_to_leading_vehicle(vehicle2, 3.0)
            vehicles.append(vehicle2)

        # 确保车辆生成
        time.sleep(5)
.........
        # 设置最大仿真时间为 16 秒
        MAX_SIM_TIME = 16

        start_time = time.time()  # 记录仿真开始时间

# 主循环
        while True:
    # 每次迭代，检查是否超过最大仿真时间
            elapsed_time = time.time() - start_time  # 计算已过去的时间
            if elapsed_time > MAX_SIM_TIME:
                break  # 超过 16 秒时退出循环

    # 进行仿真并保存数据
            world.tick()  # 获取新的帧

    # 保存GPS数据
            save_gps_data()

    # 如果有其他传感器数据需要保存，可以在这里调用对应的保存函数

    finally:
        # 清理和恢复原始设置
        world.apply_settings(original_settings)
        lidar_sensor.destroy()
        camera_sensor.destroy()
`
this are my main code but when i run it, the car will move with strange behavior such as Flashes and twitches， and it will shut down my computer and i have to restart my computer.
so what can i do for the badly issue? pleaseee , thank you alot

PatrickPromitzer, 2025-01-27:
For testing purpose you could remove the traffic_manager, spawn only one vehicle, and attach the sensor the the vehicle.
You could exclude the vehicle and only spawn the sensor directly int he world, too.

The flashes and twitches should not happen. I don't know if the save_gps_data() function does update the vehicle data, but even than is sounds like a strange behavior.

If it shut down your computer, it looks like you should test your hardware or check if your hardware is powerful enough for Carla.