carla-simulator/carla #8605: CARLA 0.9.15 : No rule to make target 'CarlaUE4Editor'.  Stop.

iasonastsaousis, 2025-01-24:
CARLA 0.9.15, source build, Ubuntu 22.04

Hello, I'm trying to run make launch, after running make PythonAPI, in order to open Unreal Engine 4 and edit the server side of my fork of CARLA, but when i do so, i get the error:

BuildCarlaUE4.sh: Using Unreal Engine at '/mnt/sda/workspace/UnrealEngine_4.26'
BuildCarlaUE4.sh: Generate Unreal project files.
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
	updated UE4Printers.py
	found necessary entries in ~/.gdbinit file, not changing it.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Discovering modules, targets and source code for project...
Generating data for project indexing...    
**Houdini Engine : Houdini 19.5.612 could not be found. Houdini Engine will not be available in this build.**
Generating data for project indexing... 100%
Writing project files... 100%
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/mnt/sda/workspace/myforkUE4/Unreal/CarlaUE4'
**make[1]: *** No rule to make target 'CarlaUE4Editor'.  Stop.**
make[1]: Leaving directory '/mnt/sda/workspace/myforkUE4/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 2

Also, everytime I run make launch, the /Unreal/CarlaUE4/.ignore file changes and I don't quite understand what this file is used for. 

From what I understand there should've been a Makefile inside the directory myforkUE4/Unreal/CarlaUE4/ with a target , but it is not there. 

**Here's everything I've tried:**

-**-> Deleting the repository and cloning it again, deleting the engine and downloading it again, the bug persists.**

**--> Reinstalling dependencies written in https://carla.readthedocs.io/en/latest/build_linux/, won't solve it**

**--> make launch-only,** i get these error messages:

[image],

[image]

**--> Tracked down this line inside the Linux.mk makefile:**

CarlaUE4Editor: LibCarla.server.release osm2odr downloadplugins
	@${CARLA_BUILD_TOOLS_FOLDER}/BuildUE4Plugins.sh --build $(ARGS)
	@${CARLA_BUILD_TOOLS_FOLDER}/BuildCarlaUE4.sh --build $(ARGS)

Inside BuildCarlaUE4.sh, I noticed that this is the point where the build process crashes:

if [ ! -f Makefile ]; then

    # This command fails sometimes but normally we can continue anyway.
    set +e
    log "Generate Unreal project files."
    ${UE4_ROOT}/GenerateProjectFiles.sh -project="${PWD}/CarlaUE4.uproject" -game -engine -makefiles
    set -e

  fi

  log "Build CarlaUE4 project."
  make CarlaUE4Editor 

I tried running this command by myself and i got this error message:

${UE4_ROOT}/GenerateProjectFiles.sh -project="/mnt/sda/workspace/myforkUE4/Unreal/CarlaUE4/CarlaUE4.uproject" -game -engine -makefiles

Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
	updated UE4Printers.py
	found necessary entries in ~/.gdbinit file, not changing it.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
Discovering modules, targets and source code for project...
Generating data for project indexing...    
**WARNING: Library '/mnt/sda/workspace/myforkUE4/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libsqlite3.so' was not resolvable to a file when used in Module 'Carla', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library '/mnt/sda/workspace/myforkUE4/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libxerces-c.a' was not resolvable to a file when used in Module 'Carla', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library '/mnt/sda/workspace/myforkUE4/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libproj.a' was not resolvable to a file when used in Module 'Carla', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
Houdini Engine : Houdini 19.5.612 could not be found. Houdini Engine will not be available in this build.**
Generating data for project indexing... 100%
Writing project files... 100%

I've tried everything and can't find what is going on. Help would be much appreciated, before I **format my partition** xD.

iasonastsaousis, 2025-01-24:
Just found new error when running make rebuild ,

/usr/bin/ld: cannot find /mnt/sda/workspace/myforkUE4/PythonAPI/carla/dependencies/lib/libxerces-c.a: No such file or directory
/usr/bin/ld: cannot find /mnt/sda/workspace/myforkUE4/PythonAPI/carla/dependencies/lib/libproj.a: No such file or directory
/usr/bin/ld: cannot find /mnt/sda/workspace/myforkUE4/PythonAPI/carla/dependencies/lib/libsqlite3.a: No such file or directory
collect2: error: ld returned 1 exit status
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
make: *** [Util/BuildTools/Linux.mk:40: rebuild] Error 1

It seems like its a similar error to the one I found at my last attempt

PatrickPromitzer, 2025-01-24:
Hi,
are you using Unreal Engine or Carla Unreal Engine?

The missing modules error happens if your are using the official Unreal Engine, but Carla is using a forked version.
https://carla.readthedocs.io/en/latest/build_linux/#unreal-engine

iasonastsaousis, 2025-01-24:
Yes, I followed the instructions from the docs. All of this happened when I tried changing code editor from VSCode to CLion. I don't know if that is what messed it up.

iasonastsaousis, 2025-01-25:
I removed dependencies again and installed them one by one and it works.