carla-simulator/carla #8606: Make PythonAPI error gzip: stdin: not in gzip format

MasoudJTehrani, 2025-01-24:
Full error when making PythonAPI or making launch for UE4 carla.

`HTTP request sent, awaiting response... 200 OK
Length: 11534 (11K) [text/html]
Saving to: ‘boost_1_80_0.tar.gz’

boost_1_80_0.tar.gz 100%[===================>]  11.26K  --.-KB/s    in 0s      

2025-01-23 10:52:46 (38.4 MB/s) - ‘boost_1_80_0.tar.gz’ saved [11534/11534]

Setup.sh: Extracting boost for Python 3.

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 2`

MasoudJTehrani, 2025-01-24:
Thanks to Patrick Promitzer on [Carla's official Discord channel](https://discord.com/invite/8kqACuC)
This answer worked for me

open the following file
path/to/carla/Util/BuildTools/Setup.sh
Go to line 94 and change this code old:

`wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true`

new:
`wget "https://archives.boost.io/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true`

closed 2025-01-27 by MarcelPiNacy-CVC

mohitgaursv, 2025-03-11:
Even though I am replacing line number 94 old: wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
with 

new: wget "https://archives.boost.io/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
it still says: 
Setup.sh: Extracting boost for Python 3.

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 2
adlab4@adlab4-HP-Z8-G5-Workstation-Desktop-PC:~/UnrealEngine_4.26/Engine/Binaries/Linux/carla$

ssysm, 2025-05-20:
You need to remove the old failed file in the Build/ directory