carla-simulator/carla #8608: Carla 0.10.0 on UE5.5 crashes when running manual_control.py

galadeat, 2025-01-24:
[image]
Hi, Carla Community!
UE-CarlaUnreal Editor crushes, when I run manual_control.py or automatic_control.py. Other scripts run as they should (I tested generate_traffic.py).  Sometimes rebuild of the whole Carla project helps, but I would like to know the specific reason why crash may occur.

galadeat, 2025-01-28:
[CarlaUnreal.log](https://github.com/user-attachments/files/18574579/CarlaUnreal.log)

MarcelPiNacy-CVC, 2025-01-28:
Is this from a package? Or was this manually built?

galadeat, 2025-01-29:
It was manually built according to [the carla docs](https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/)

galadeat, 2025-01-31:
There is another occurrence of the problem: I tried to turn off nanite meshes show and the same problem popped up  

[image]

MarcelPiNacy-CVC, 2025-01-31:
Try switching to the branch marcel/attr-type-fixes. This may fix your issue if I haven't missed anything.

galadeat, 2025-02-03:
@MarcelPiNacy-CVC , thank you for your reply! I've rebuilt the project on your branch and now manual_control.py works. But the problem with turning off nanite meshes remains.

MarcelPiNacy-CVC, 2025-02-03:
Could you share the crash log in this case? I don't think turning nanite off is a good idea though, Town10 currently has a lot of geometry.

galadeat, 2025-02-04:
@MarcelPiNacy-CVC , here is the [CarlaUnreal.log](https://github.com/user-attachments/files/18653303/CarlaUnreal.log).