carla-simulator/carla #8611: Getting the id or name of the newly spawned prop through PythonAPI

MasoudJTehrani, 2025-01-29:
Hello everyone, 
I used the PythonAPI to spawn a prop, namely, the "[static.prop.ironplank](https://carla.readthedocs.io/en/latest/catalogue_props/#iron-plank)". How can I then get the ID or the name of the prop in my current world?

I used this code:
```
names = world.get_names_of_all_objects()
    for n in sorted(names):
        print(n)
```
and manually searched through the values and found that my prop is called "StaticMeshActor_2147479215". 
How can I do it without manually searching through the names of all objects?

This is the code that I used to spawn the prop

```
prop = blueprints.find('static.prop.ironplank')
    location = carla.Location(x=80, y=108, z=7)
    ts = world.try_spawn_actor(prop, carla.Transform(location))
```
and if I print `ts` I get: 
`Actor(id=26, type=static.prop.ironplank)`
Which is not what I am looking for

PatrickPromitzer, 2025-01-29:
Hi,
what do you want to do with that information?

If you want to spawn an actor and find the actor instance at a later time, you can use the "role_name" attribute.

```python

client: carla.Client = carla.Client("127.0.0.1", 2000)
world: carla.World = client.get_world()
blueprints = world.get_blueprint_library()
prop = blueprints.find("static.prop.ironplank")

prop.set_attribute('role_name', 'hero')
ts = world.try_spawn_actor(prop, carla.Transform(location))

# at  later time

actor_list = world.get_actors()

your_actor = None
for actor in actor_list:
    if "role_name" not in actor.attributes:
        continue
    actor_role: str = actor.attributes["role_name"]
    if actor_role == "hero":
        your_actor = actor
```

It used role_name with vehicles, but it should work with all actors.

MasoudJTehrani, 2025-01-30:
Hi,
I want to then change its texture using
`world.apply_color_texture_to_object('StaticMeshActor_2147479217', carla.MaterialParameter.Diffuse, texture)`
and unfortunately, your solution is not applicable in this case.
I guess I have to either use UE editor to change its name or just look for a `StaticMeshActor` that has the longest number in front of it.

PatrickPromitzer, 2025-01-30:
Hi,
you could look in the actor.attributes if the object name is included, or maybe it is somewhere else in the actor.
I would need to check the actor class myself to be sure.