carla-simulator/carla #8613: The version issue of CARLA.

xss1006, 2025-01-29:
Does the version of CARLA have a significant impact on the training effectiveness of the algorithm?

MarcelPiNacy-CVC, 2025-01-30:
Hi, are you asking whether it performs better?

xss1006, 2025-01-31:
> 你好，你问的是它的性能是否更好？

Thank you for your reply.The question I want to ask is whether different versions of CARLA have an impact on the training performance of reinforcement learning algorithms.

PabloVD, 2025-01-31:
It depends on the use you require for your training. Carla 0.10 have better camera fidelity with more realistic graphics and illumination with respect to previous versions, which may have an impact in the results. Note also that 0.10 only includes two maps, where previous releases include more, which would be better for variability. Weather is also disabled in 0.10, so 0.9.x can provide more variable scenarios within the same map varying the weather conditions. Besides that, check out [here](https://carla.org/2024/12/19/release-0.10.0/) the rest of features the new version includes and some others that are not included.

closed 2025-04-01 by Blyron