carla-simulator/carla #8614: Cannot launch Carla UE5 after carla api install on Windows 11

granddrumsco, 2025-01-30:
The problem same as #8594 is occurred.
 "launch" does not exist in target list

c:\carla\CarlaUE5>cmake --build Build --target help
[0/2] Re-checking globbed directories...
[1/1] All primary targets available:
carla-help: phony
edit_cache: phony
rebuild_cache: phony
list_install_components: phony
install: phony
install/local: phony
C:/carla/CarlaUE5/Build/CMakeFiles/carla-help: CUSTOM_COMMAND
_deps/zlib-build/zlib.lib: C_SHARED_LIBRARY_LINKER__zlib_Release
_deps/zlib-build/edit_cache: phony
_deps/zlib-build/rebuild_cache: phony
_deps/zlib-build/list_install_components: phony
_deps/zlib-build/install: phony
_deps/zlib-build/install/local: phony
_deps/libpng-build/edit_cache: phony
_deps/libpng-build/rebuild_cache: phony
_deps/libpng-build/list_install_components: phony
_deps/libpng-build/install: phony
_deps/libpng-build/install/local: phony
_deps/boost-build/edit_cache: phony
_deps/boost-build/rebuild_cache: phony
_deps/boost-build/list_install_components: phony
_deps/boost-build/install: phony
_deps/boost-build/install/local: phony
_deps/boost-build/libs/algorithm/edit_cache: phony
_deps/boost-build/libs/algorithm/rebuild_cache: phony
_deps/boost-build/libs/algorithm/list_install_components: phony
_deps/boost-build/libs/algorithm/install: phony
_deps/boost-build/libs/algorithm/install/local: phony
_deps/boost-build/libs/align/edit_cache: phony
_deps/boost-build/libs/align/rebuild_cache: phony
_deps/boost-build/libs/align/list_install_components: phony
_deps/boost-build/libs/align/install: phony
_deps/boost-build/libs/align/install/local: phony
_deps/boost-build/libs/any/edit_cache: phony
_deps/boost-build/libs/any/rebuild_cache: phony
_deps/boost-build/libs/any/list_install_components: phony
_deps/boost-build/libs/any/install: phony
_deps/boost-build/libs/any/install/local: phony
_deps/boost-build/libs/array/edit_cache: phony
_deps/boost-build/libs/array/rebuild_cache: phony
_deps/boost-build/libs/array/list_install_components: phony
_deps/boost-build/libs/array/install: phony
_deps/boost-build/libs/array/install/local: phony
_deps/boost-build/libs/asio/edit_cache: phony
_deps/boost-build/libs/asio/rebuild_cache: phony
_deps/boost-build/libs/asio/list_install_components: phony
_deps/boost-build/libs/asio/install: phony
_deps/boost-build/libs/asio/install/local: phony
_deps/boost-build/libs/assert/edit_cache: phony
_deps/boost-build/libs/assert/rebuild_cache: phony
_deps/boost-build/libs/assert/list_install_components: phony
_deps/boost-build/libs/assert/install: phony
_deps/boost-build/libs/assert/install/local: phony
_deps/boost-build/libs/atomic/edit_cache: phony
_deps/boost-build/libs/atomic/rebuild_cache: phony
_deps/boost-build/libs/atomic/list_install_components: phony
_deps/boost-build/libs/atomic/install: phony
_deps/boost-build/libs/atomic/install/local: phony
_deps/boost-build/libs/bimap/edit_cache: phony
_deps/boost-build/libs/bimap/rebuild_cache: phony
_deps/boost-build/libs/bimap/list_install_components: phony
_deps/boost-build/libs/bimap/install: phony
_deps/boost-build/libs/bimap/install/local: phony
_deps/boost-build/libs/bind/edit_cache: phony
[... 524 lines cut ...]
libboost_date_time-vc143-mt-x64-1_84.lib: phony
libboost_exception-vc143-mt-x64-1_84.lib: phony
libboost_filesystem-vc143-mt-x64-1_84.lib: phony
libboost_graph-vc143-mt-x64-1_84.lib: phony
libboost_numpy310-vc143-mt-x64-1_84.lib: phony
libboost_python310-vc143-mt-x64-1_84.lib: phony
libboost_random-vc143-mt-x64-1_84.lib: phony
libboost_serialization-vc143-mt-x64-1_84.lib: phony
libboost_thread-vc143-mt-x64-1_84.lib: phony
libboost_wserialization-vc143-mt-x64-1_84.lib: phony
libpng16_static.lib: phony
libsqlite3: phony
png_genfiles: phony
png_static: phony
rpc: phony
rpc.lib: phony
sqlite3: phony
uninstall: phony
zlib: phony
zlib.dll: phony
zlibstatic: phony
zlibstatic.lib: phony
build.ninja: RERUN_CMAKE
clean: CLEAN
help: HELP

MarcelPiNacy-CVC, 2025-01-31:
Did you manually build CARLA here? It looks like the unreal targets are disabled.

granddrumsco, 2025-02-03:
Following the description "Build and install the Python API. Open the x64 Native Tools Command Prompt for VS 2022 at the CarlaUE5 folder and run the following command:", I succeeded CARLA Python API build+install.

And I tried "Launch the editor" with following the description "Open the x64 Native Tools Command Prompt for VS 2022 at the CarlaUE5 folder and run the following command:".

C:\carla\CarlaUE5>cmake --build Build --target launch
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?

I checked target list, but I could not find out "launch".
I would like to know how to launch the editor.

dominikmaeckel, 2025-02-04:
> Following the description "Build and install the Python API. Open the x64 Native Tools Command Prompt for VS 2022 at the CarlaUE5 folder and run the following command:", I succeeded CARLA Python API [...]

I am having exactly the same issue as @granddrumsco.

granddrumsco, 2025-02-06:
Thank you for your support.
I did your suggestion.

C:\carla\CarlaUE5>git clean -fxd
Removing Build/
Removing LibCarla/source/carla/Version.h
Removing LibCarla/source/util/ue-header-guard-begin.h
Removing LibCarla/source/util/ue-header-guard-end.h
Removing PythonAPI/carla/CMakeLists.txt
Removing PythonAPI/carla/LICENSE
Removing PythonAPI/carla/pyproject.toml
Removing Temp/

But my situation is same as before.

C:\carla\CarlaUE5>CarlaSetup.bat
・
・
・
CARLA Python API build+install succeeded.
Launching Carla Unreal Editor...
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?

My environment is as follows.
Windows11
Microsoft Visual Studio Community 2022 (64-bit) - Current Version 17.12.3
Python3.10.11

MarcelPiNacy-CVC, 2025-02-06:
In this case, try using the following commands:
```
git clean -fxd
cmake --preset Development
cmake --build Build/Development
cmake --build Build/Development --target carla-python-api-install
cmake --build Build/Development --target launch
```
Notice that this changes your build folder from Build/ to Build/Development.

granddrumsco, 2025-02-07:
I tried the above commands, but my situation did not change.
C:\carla\CarlaUE5>git clean -fxd
・
・
・
C:\carla\CarlaUE5>git pull
・
・
・
C:\carla\CarlaUE5>cmake --preset Development
Preset CMake variables:

  CMAKE_BUILD_TYPE="RelWithDebInfo"
  CMAKE_EXPORT_COMPILE_COMMANDS:BOOL="TRUE"
  CMAKE_INSTALL_PREFIX:PATH="C:/carla/CarlaUE5/Install/Development"
  CMAKE_TOOLCHAIN_FILE="C:/carla/CarlaUE5/CMake/Toolchain.cmake"
・
・
・
C:\carla\CarlaUE5>cmake --build Build/Development
・
・
・[414/414] Linking CXX executable Examples\carla-example-client.exe
C:\carla\CarlaUE5>cmake --build Build/Development --target carla-python-api-install
・
・
・
Successfully installed carla-0.10.0
C:\carla\CarlaUE5>cmake --build Build/Development --target launch
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?

MarcelPiNacy-CVC, 2025-02-07:
Please share the full output log from these commands.

granddrumsco, 2025-02-10:
I did your suggestion again today.
The full output log is the followings.

C:\carla\CarlaUE5>git clean -fxd
Removing .clangd
Removing Build/
Removing LibCarla/source/carla/Version.h
Removing LibCarla/source/util/ue-header-guard-begin.h
Removing LibCarla/source/util/ue-header-guard-end.h
Removing PythonAPI/carla/CMakeLists.txt
Removing PythonAPI/carla/LICENSE
Removing PythonAPI/carla/pyproject.toml

C:\carla\CarlaUE5>git pull
remote: Enumerating objects: 31, done.
remote: Counting objects: 100% (31/31), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 31 (delta 24), reused 30 (delta 24), pack-reused 0 (from 0)
Unpacking objects: 100% (31/31), 7.04 KiB | 35.00 KiB/s, done.
From https://github.com/carla-simulator/carla
 * [new branch]          jorge/ue5/bone-transform-mismatch-fix -> origin/jorge/ue5/bone-transform-mismatch-fix
   bf114a122..8e9d0c4ce  ue4-dev    -> origin/ue4-dev
Already up to date.

C:\carla\CarlaUE5>cmake --preset Development
Preset CMake variables:

  CMAKE_BUILD_TYPE="RelWithDebInfo"
  CMAKE_EXPORT_COMPILE_COMMANDS:BOOL="TRUE"
  CMAKE_INSTALL_PREFIX:PATH="C:/carla/CarlaUE5/Install/Development"
  CMAKE_TOOLCHAIN_FILE="C:/carla/CarlaUE5/CMake/Toolchain.cmake"

-- The C compiler identification is MSVC 19.38.33143.0
-- The CXX compiler identification is MSVC 19.38.33143.0
-- The ASM compiler identification is MSVC
-- Found assembler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CARLA: Using  as Unreal Engine root path.
-- Performing Test HAS_MSVC_UTF8
-- Performing Test HAS_MSVC_UTF8 - Success
-- CARLA: Checking for /EHsc support
-- Performing Test HAS_EXCEPTIONS_FLAG
-- Performing Test HAS_EXCEPTIONS_FLAG - Success
-- CARLA: Checking for /GR support
-- Performing Test HAS_RTTI_FLAG
-- Performing Test HAS_RTTI_FLAG - Success
-- Performing Test HAS_macro_redefined
-- Performing Test HAS_macro_redefined - Failed
-- Performing Test HAS__4005
-- Performing Test HAS__4005 - Success
-- Performing Test HAS_incompatible_pointer_types
-- Performing Test HAS_incompatible_pointer_types - Failed
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
-- CARLA: Downloading sqlite3...
-- CARLA: Downloading zlib...
[... 225 lines cut ...]
-- Install configuration: "RelWithDebInfo"
-- Installing: C:\Users\XXXXX\AppData\Local\Temp\tmp3yheh4sb\wheel\platlib/./carla.cp310-win_amd64.pyd
*** Making wheel...
*** Created carla-0.10.0-cp310-cp310-win_amd64.whl
Successfully built carla-0.10.0-cp310-cp310-win_amd64.whl

[1/2] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\carla\Carla...n.exe -m pip install -e C:/carla/CarlaUE5/PythonAPI/carla"Obtaining file:///C:/carla/CarlaUE5/PythonAPI/carla
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Building wheels for collected packages: carla
  Building editable for carla (pyproject.toml) ... done
  Created wheel for carla: filename=carla-0.10.0-cp310-cp310-win_amd64.whl size=3748559 sha256=bb2ffe2cbb1fae3d90dc2d26d2e0f9ffbaf4f47c9347a3e3cf5f8672a8ba7e70
  Stored in directory: C:\Users\XXXXX\AppData\Local\Temp\pip-ephem-wheel-cache-9khvji4t\wheels\6c\05\8c\859d9c96d6604ff8834fd467374f3bdd7e1ba9d4dfcff1867b
Successfully built carla
Installing collected packages: carla
  Attempting uninstall: carla
    Found existing installation: carla 0.10.0
    Uninstalling carla-0.10.0:
      Successfully uninstalled carla-0.10.0
Successfully installed carla-0.10.0

C:\carla\CarlaUE5>cmake --build Build/Development --target launch
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?

granddrumsco, 2025-03-04:
I tried various recommendations searching web.
Finally I could run "cmake --build Build --target launch" command.
I think that it is good for me to modify Visual Studio 2022 configuration MSVC v143 - VS 2022 C++ x64/x86 Build Tool(V14.38 - 27.8) enabled.

closed 2025-03-04 by granddrumsco

xiao1874, 2025-03-06:
> I tried various recommendations searching web. Finally I could run "cmake --build Build --target launch" command. I think that it is good for me to modify Visual Studio 2022 configuration MSVC v143 - VS 2022 C++ x64/x86 Build Tool(V14.38 - 27.8) enabled.

Hi, I encountered the same problem as you. Can you describe in detail how you solved it? Thank you!

xiao1874, 2025-03-06:
> MSVC v143 - VS 2022 C++ x64/x86 Build Tool(V14.38 - 27.8)
In Visual Studio Installer, I only find MSVC v143 - VS 2022 C++ x64/x86 Build Tool(V14.38 - 17.8)

xiao1874, 2025-03-06:
> > I tried various recommendations searching web. Finally I could run "cmake --build Build --target launch" command. I think that it is good for me to modify Visual Studio 2022 configuration MSVC v143 - VS 2022 C++ x64/x86 Build Tool(V14.38 - 27.8) enabled.
> 
> Can you specify what you did exactly?

In Visual Studio Installer, I only find MSVC v143 - VS 2022 C++ x64/x86 Build Tool(**V14.38 - 17.8**)

granddrumsco, 2025-03-07:
I'm sorry.
MSVC v143 - VS 2022 C++ x64/x86 Build Tool(V14.38 - 17.8) is correct.
I could not specify the cause of problem.

ajay1606, 2025-03-12:
@granddrumsco Thanks for sharing this, great you fixed it finally.

I am also trying to do same, but unfortunately unable to succeed. 

[image]

```
C:\CarlaUE5>cmake --build Build --target launch
[0/2] Re-checking globbed directories...ninja: error: unknown target 'launch', did you mean 'lapack'?
```
Getting error as above even after choosing correct MSVC version, any suggestion please.

Appreciate any help

dedoogong, 2025-03-29:
I met the same problem! please help me~!

aaadam3042, 2025-04-30:
After digging around I have discovered the issue appears to have to do with cmake failing to add the` \unreal` directory in the root as a subdirectory, which is set in the root CMakeLists.txt. Trying to print the BUILD_CARLA_UNREAL option seems to always return OFF as well as UNREAL_ENGINE_PATH appearing empty. This is despite `cmake --build BUILD --target carla-help` reporting that those options are set correctly.

Temporary work around is to hardcode in the root CMakeLists.txt file:
`set (BUILD_CARLA_UNREAL ON)`
somewhere before the line containing `if (BUILD_CARLA_UNREAL)`

and then rerun:
`cmake -G Ninja -S . -B Build -DCMAKE_BUILD_TYPE=Release`

aaadam3042, 2025-04-30:
Problematic code seems to be:
`if (EXISTS ${CARLA_UNREAL_ENGINE_PATH})
  set (CARLA_HAS_UNREAL_ENGINE_PATH ON)
else ()
  set (CARLA_HAS_UNREAL_ENGINE_PATH OFF)
endif ()

carla_option (
  BUILD_CARLA_UNREAL
  "Build the Carla Unreal Engine project."
  ${CARLA_HAS_UNREAL_ENGINE_PATH}
)`

adding `set (BUILD_CARLA_UNREAL ON)` and `set (BUILD_CARLA_UNREAL OFF)` to respective branches fixes the issue.

If I'm also not mistaken we can remove the CARLA_HAS_UNREAL_ENGINE_PATH flag entirely as the only other place its used is further down nested in:
`if (BUILD_CARLA_UNREAL)
  if (NOT ${CARLA_HAS_UNREAL_ENGINE_PATH})`
which doesn't serve a real purpose anyway as we were initially trying to set BUILD_CARLA_UNREAL with CARLA_HAS_UNREAL_ENGINE_PATH

Shunqiang-Feng, 2025-10-19:
> Problematic code seems to be:
> 
> ```
> if (EXISTS ${CARLA_UNREAL_ENGINE_PATH})
>   set (CARLA_HAS_UNREAL_ENGINE_PATH ON)
> else ()
>   set (CARLA_HAS_UNREAL_ENGINE_PATH OFF)
> endif ()
> 
> carla_option (
>   BUILD_CARLA_UNREAL
>   "Build the Carla Unreal Engine project."
>   ${CARLA_HAS_UNREAL_ENGINE_PATH}
> )
> ```
> 
> adding and to respective branches fixes the issue.`set (BUILD_CARLA_UNREAL ON)``set (BUILD_CARLA_UNREAL OFF)`
> 
> If I'm also not mistaken we can remove the CARLA_HAS_UNREAL_ENGINE_PATH flag entirely as the only other place its used is further down nested in: which doesn't serve a real purpose anyway as we were initially trying to set BUILD_CARLA_UNREAL with CARLA_HAS_UNREAL_ENGINE_PATH`if (BUILD_CARLA_UNREAL) if (NOT ${CARLA_HAS_UNREAL_ENGINE_PATH})`
> 
> In summary the issue appears to be that ${CARLA_HAS_UNREAL_ENGINE_PATH} is evaluating in carla_option after the whole cmake build has completed running and thus can't be used to set the value of BUILD_CARLA_UNREAL within the same cmake command, at least on windows 11 with cmake version 4.0.1

No, CMake < 3.5 is required.

CMake Error at Build/_deps/libpng-src/CMakeLists.txt:33 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.