carla-simulator/carla #8624: Create custom maps in CARLA 0.10.0

Cabba, 2025-01-31:
Hi,

I'm trying to create a custom map for CARLA 0.10.0 and I'm running into some issues. I was looking at the online documentation, specifically this section: [https://carla-ue5.readthedocs.io/en/latest/maps_tutorials/](https://carla-ue5.readthedocs.io/en/latest/maps_tutorials/)

Unfortunately, all the links within this section appear to be broken, making it impossible for me to follow the tutorial.

Therefore, I have the following questions:

- **What is the recommended procedure for building a new CARLA map from scratch using the latest CARLA version (0.10.0)?**  I'm looking for the most up-to-date and efficient workflow.  Any guidance, links to tutorials, or example files would be greatly appreciated.

- **Is RoadRunner compatible with the latest CARLA release (0.10.0)?** If so, are there any specific considerations or best practices I should be aware of when using RoadRunner to create maps for CARLA 0.10.0?

Thank you for your assistance.

PatrickPromitzer, 2025-02-03:
Hi,
the broken links comes from the documentation of the UE4 version. The UE5 documentation is still work-in-process.
https://carla.readthedocs.io/en/latest/maps_tutorials/

Currently, there is no procedure for building a custom Carla map with Carla 0.10.0
I have ideas how you can create an custom Carla map, but you don't have an import script, template map, or other parts to give you a working base map with a few simple steps.

The files RoadRunner is exporting are 
- openDRIVE: Road data 
- Filmbox: Model data (objects, texture, ...)

Unreal Engine is supporting *.fbx (Filmbox) files,and therefor still work with the files from RoadRunner.

If you want to create an custom Map and don't want to wait, you could do the following.
- Make a copy of the working Town10 map
- rename the copy to a custom name (your custom map)
- remove all actors/objects from the custom map (make sure to not delete non object files like sky)
- import the .fbx file in Unreal Engine
- add the map object of the .fbx file in your custom map 

I didn't try if it works, but it would be worth a shot,

timmao955, 2025-02-06:
I tried to build a new map on CARLA 0.10.0 but failed. Unfortunately, it seems like that CARLA 0.10.0 doesn't support to build additional maps package. Instead, I find some methods, which might be a bit complicated.

- As the above said, copy the Town10HD_Opt map in CARLA 0.10.0 and remove most of the blueprints
- Use the RoadRunner to export the .xodr file, and there's a plugin [https://github.com/brifsttar/OpenDRIVE.git](url) that can help you to project the .xodr net in UE
- You can create your custom map with additional blueprints (or using .fbx file) based on this plugin in UE
- Don't forget to reserve some default actors in CARLA 0.10.0, like VehicleSpawnPoint, OpenDriveActor, e.g.

I tried this method and it works, but it's still under development

Blyron, 2026-11-12:
Hi, sorry for the late answer, and thanks to @PatrickPromitzer and @junhengm1 for the workarounds.

### CARLA 0.10.0

0.10.0 has no supported workflow for building a custom map: no import script, no template map, and no way to package a map for an existing CARLA build. The options there are:

- **OpenDRIVE standalone mode** if you only need roads. 0.10.0 ships `OpenDriveMap`, so you can load any `.xodr` at runtime without the editor. CARLA generates the road mesh, and you get no scenery:
  ```python
  with open("my_map.xodr") as f:
      client.generate_opendrive_world(f.read())
  ```
- **The manual approach described above** if you need the scenery: duplicate Town10HD_Opt in a source build, strip it, then add your FBX and the `.xodr`. Keep CARLA's own actors in the map (e.g. `OpenDriveActor` and the vehicle spawn points). This is not supported, it only works inside that source build, and it is easy to break traffic, pedestrians or packaging.

To answer your questions for this release: RoadRunner's `.fbx` + `.xodr` export loads in Unreal 5.5, but 0.10.0 has nothing to import it as a CARLA map. Any tool that writes valid ASAM OpenDRIVE works for standalone mode.

### `ue58-dev` / CARLA 1.0 (UE 5.8)

We are migrating to `ue58-dev`, which becomes CARLA 1.0, and custom maps are supported there:

1. **Create the road network** in [RoadRunner](https://carla-ue5.readthedocs.io/en/latest/roadrunner/), which is the recommended editor. That page covers the export settings CARLA expects. TrueVision Designer produces the same pair of files if you do not have a RoadRunner licence.
2. **Import** the `.fbx` + `.xodr` with `python3 Util/Tools/Import.py --package <name>`. It imports the geometry, applies CARLA's road materials, prepares the map for cooking and builds the Traffic Manager cache. See [Content authoring - maps](https://carla-ue5.readthedocs.io/en/latest/content_authoring_maps/).
3. **Decorate** the map in the editor. Assets from Fab or elsewhere can be added as meshes in the map, or as spawnable props. See [Content authoring - props](https://carla-ue5.readthedocs.io/en/latest/content_authoring_props/).
4. **Ship** the map as a [content pack](https://carla-ue5.readthedocs.io/en/latest/content_authoring_packs/), which installs into any CARLA package of the same release without rebuilding CARLA.

[Custom maps](https://carla-ue5.readthedocs.io/en/latest/content_authoring_maps_overview/) compares all the routes, including [OpenDRIVE standalone mode](https://carla-ue5.readthedocs.io/en/latest/core_opendrive/). The `carla-ue5.readthedocs.io/en/latest` docs now follow `ue58-dev`, and that page replaces the `maps_tutorials` page you linked.

closed 2026-11-12 by Blyron