carla-simulator/carla #8626: Linux Carla build errorr

AKG47-Auto, 2025-01-31:
[log.txt](https://github.com/user-attachments/files/18624780/log.txt)
Carla 0.9.15 build on Linux 22.04 
got the following error: 
fatal error: too many errors emitted, stopping now [-ferror-limit=] 20 errors generated. ninja: build stopped: subcommand failed. make: *** [Util/BuildTools/Linux.mk:157: osm2odr] Error 1: 
Can anyone help me with this??
If someone can check the logs and help resolve the issue. Thanks

MarcelPiNacy-CVC, 2025-01-31:
Hi,
Do you have CARLA's UE4 fork downloaded and built already?

AKG47-Auto, 2025-01-31:
> Hi, Do you have CARLA's UE4 fork downloaded and built already?

Yes the UE4.26 was built successfully and was running. I followed the official readme docs for the build.

PatrickPromitzer, 2025-02-03:
Hi,
can you try if "make launch" works?

AKG47-Auto, 2025-02-06:
> Hi, can you try if "make launch" works?

Tried that too. same error