carla-simulator/carla #8628: Handles occluded objects

KJ-Chang, 2025-02-04:
I find all the objects in the world and use the Bounding Box to frame the 2D Bounding Box of the objects in the field of view.

But I'm having some problems, when the object is obscured, I still draw his 2D Bounding Box, which is obviously wrong, can someone tell me how to determine if the object is occluded and don't draw a 2D Bounding Box? 

[image]

[image]

Here's my code

```python
def show_radar_points_on_camera(world: World):
    rgb_image_data = rgb_cam_queue.get()
    rgb_img = np.copy(np.frombuffer(rgb_image_data.raw_data, dtype=np.dtype('uint8')))
    rgb_img = np.reshape(rgb_img, (rgb_image_data.height, rgb_image_data.width, 4))
    
    radar_data = rdr_queue.get()
    rdr_size = len(radar_data)
    rdr = np.copy(np.frombuffer(radar_data.raw_data, dtype=np.dtype('f4')))
    rdr = np.reshape(rdr, (rdr_size, 4))

    # 確保
    # if rgb_image_data.frame != depth_image_data.frame or depth_image_data.frame!= radar_data.frame or radar_data.frame != rgb_image_data.frame:
    #     raise ValueError(f"rgb_frame:{rgb_image_data.frame}  depth_frame:{depth_image_data.frame}  radar_frame:{radar_data.frame} not exactly the same!!!")

    # Collect all 
    npcs = []
    npc_filter_list = ["vehicle.*", "walker.pedestrian.*",]
    for f in npc_filter_list:
        for npc in world.world.get_actors().filter(f):
            npcs.append(npc)
    
    city_object_labels = [carla.CityObjectLabel.Pedestrians, 
                          carla.CityObjectLabel.Car, 
                          carla.CityObjectLabel.Bus, 
                          carla.CityObjectLabel.Truck, 
                          carla.CityObjectLabel.Motorcycle, 
                          carla.CityObjectLabel.Bicycle, 
                          carla.CityObjectLabel.Rider,
                          ]
    for l in city_object_labels:
        for npc in world.world.get_environment_objects(object_type=l):
            npcs.append(npc)
   
    # Show 2D Bounding Box
    my_vehicle = world.player
    for npc in npcs:

        x_max = -10000
        x_min = 10000
        y_max = -10000
        y_min = 10000

        cam_right = g_camera_sensor.get_transform().get_right_vector()
        cam_forward_vec = g_camera_sensor.get_transform().get_forward_vector()
    
        adjust_pixel_point = []

        bbox_color = (255, 255, 255)
        # 動態物件(用藍圖產生)
        if type(npc) in [carla.libcarla.Vehicle, carla.libcarla.Walker]:
            # 排除自己
            if npc.id == my_vehicle.id:
                continue

            if npc.type_id.split('.')[0] == 'vehicle':
                bbox_color = (0, 0, 255)
            else:
                bbox_color = (255, 0, 0)

[... 146 lines cut ...]
            cv2.line(rgb_img, (round(x_min),round(y_min)), (round(x_max),round(y_min)), bbox_color, 2)
            cv2.line(rgb_img, (round(x_min),round(y_max)), (round(x_max),round(y_max)), bbox_color, 2)
            cv2.line(rgb_img, (round(x_min),round(y_min)), (round(x_min),round(y_max)), bbox_color, 2)
            cv2.line(rgb_img, (round(x_max),round(y_min)), (round(x_max),round(y_max)), bbox_color, 2)
            # cv2.circle(rgb_img, (round(bbox_center_x), round(bbox_center_y)), radius=2, color=(0, 255, 0))
                          
                    
    
    # (vel, azimuth, altitude, depth)    但是官方範例是寫成->(vel, altitude, azimuth, depth) 官方給的畫出來的點會是錯的
    # for (_, azimuth, altitude, depth) in rdr:
    #     # print(f'仰角:{math.degrees(altitude)}, 方位角:{math.degrees(azimuth)}, 深度:{depth}')
    #     radar_local = get_radar_local(altitude, azimuth, depth)
    #     radar_matrix = g_radar_sensor.get_transform().get_matrix()
    #     radar_local_to_world = get_radar_local_to_world(radar_local, radar_matrix)
    #     camera_inverse_matrix = g_camera_sensor.get_transform().get_inverse_matrix()
    #     camera_world_to_local = get_camera_world_to_local(radar_local_to_world, camera_inverse_matrix)
    #     image_point = get_image_point(camera_world_to_local, K)
    #     if (image_point[0]<0) | (image_point[0]>=rgb_image_data.width) | (image_point[1]<0) | (image_point[1]>=rgb_image_data.height):
    #         continue
    #     cv2.circle(rgb_img, center=(round(image_point[0]), round(image_point[1])), radius=2, color=(0, 0, 255), thickness=-1)
    #     cv2.putText(img=rgb_img, text=str(int(depth)), org=(int(image_point[0])-1, int(image_point[1])), color=(255, 0, 0), fontFace=cv2.FONT_HERSHEY_SIMPLEX, fontScale=0.35, lineType=cv2.LINE_AA)
    
    cv2.imshow("ImageWindowName", rgb_img)
    cv2.waitKey(1)
```

GoodarzMehr, 2025-02-07:
One thing you could try is to add a depth camera with the same properties as your RGB camera and use its output to filter out occluded bounding boxes.

KJ-Chang, 2025-02-08:
> One thing you could try is to add a depth camera with the same properties as your RGB camera and use its output to filter out occluded bounding boxes.

I'll try to do it, thank you for your reply!