carla-simulator/carla #8640: CarlaSetup.sh options to be corrected

abhishekkumardwivedi, 2025-02-06:
Currently the options in CarlaSetup.sh is like:

```
while true; do
    case "$1" in
        -i|--interactive)
            interactive=1
            shift
            ;;
        -p|--skip-prerequisites)
            skip_prerequisites=1
            shift
            ;;
        -l|--launch)
            launch=1
            shift
            ;;
        -pyroot|--python-root)
            python_root=$2
            shift 2
            ;;
        --)
            shift
            break
            ;;
        *)
            ;;
    esac
done
```
Here current logic excepts passed arguments as first argument i.e. $1, which seems causing issue. As at a time we can pass it as interactive then we can not do for launch. A better design would be to not have all static order rather look for what options are passed and do corresponding swith operations recursively. Or pre-define seperate argument positions for launch and interactive per say.

MarcelPiNacy-CVC, 2025-02-06:
Hi, I'm not sure I follow what the problem is.

Blyron, 2025-02-20:
Explain the issue better please, otherwise we will be closing the issue

closed 2025-02-24 by MarcelPiNacy-CVC