carla-simulator/carla #8645: Make Python API error in dev-ue5 branch code

donghaiwang, 2025-02-07:
Launch Editor success. However, there are some error when execute these command (Windows 10, Python 12):
```
cmake --build Build --target carla-python-api-install
```

Error information: 
```
(ue5-dev) C:\workspace\CarlaUE5>cmake --build Build --target carla-python-api-install
[0/2] Re-checking globbed directories...
[0/2] C:\Windows\system32\cmd.exe /C "cd /D C:\workspace\C...outdir C:/workspace/CarlaUE5/Build/PythonAPI/dist --wheel"* Creating isolated environment: venv+pip...
* Installing packages in isolated environment:
  - scikit-build-core
  - wheel
* Getting build dependencies for wheel...
* Building wheel...
*** scikit-build-core 0.10.7 using CMake 3.29.5 (wheel)
*** Configuring CMake...
2025-02-07 13:09:46,090 - scikit_build_core - WARNING - Can't find a Python library, got libdir=None, ldlibrary=None, multiarch=None, masd=None
loading initial cache file C:\Users\Administrator\AppData\Local\Temp\tmp0x4rae7o\build\CMakeInit.txt
CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.29/Modules/CMakeDetermineSystem.cmake:152 (message):
  Could not find toolchain file:
  C:/workspace/CarlaUE5/$PWD/CMake/LinuxToolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

*** CMake configuration failed

ERROR Backend subprocess exited when trying to invoke build_wheel

FAILED: PythonAPI/CMakeFiles/carla-python-api C:/workspace/CarlaUE5/Build/PythonAPI/CMakeFiles/carla-python-api
C:\Windows\system32\cmd.exe /C "cd /D C:\workspace\CarlaUE5\PythonAPI\carla && "C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E copy C:/workspace/CarlaUE5/LICENSE C:/workspace/CarlaUE5/PythonAPI/carla/LICENSE && C:\software\anaconda3\envs\ue5-dev\python.exe -m build --outdir C:/workspace/CarlaUE5/Build/PythonAPI/dist --wheel"
ninja: build stopped: subcommand failed.

```

MarcelPiNacy-CVC, 2025-02-07:
Hi, I recently renamed the LinuxToolchain for several reasons. You will need to rebuild CARLA, I'd suggest you run `git clean -fxd` before rerunning the build steps.

MarcelPiNacy-CVC, 2025-02-10:
Hi, I'm closing this for now. Feel free to reopen it if this still fails.

closed 2025-02-10 by MarcelPiNacy-CVC