carla-simulator/carla #8646: Can't find CarlaUnreal.sh after building

ihshin227, 2025-02-07:
Hello, I followed this document and build was succeeded.
https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/

But I couldn't find the CarlaUnreal.sh file in my CarlaUE5 directory.

MarcelPiNacy-CVC, 2025-02-07:
The CarlaUnreal.sh is generated when building a package. To launch use `cmake --build Build --target launch`.

MarcelPiNacy-CVC, 2025-02-07:
I'm closing this for now, feel free to reopen it if you need to.

ihshin227, 2025-02-07:
Thank you for reply.

When I use `cmake --build Build --target launch` to generate the CarlaUnreal.sh, I met a `Signal 11 caught.` message.

[image]

my environment is followed as:
- OS: Ubuntu 22.04
- CPU: AMD Ryzen Threadripper 3970x 32-Core
- GPU: NVIDIA GeForce RTX 3090
  - Driver Version: 550.144.03
  - CUDA Version: 12.4
- RAM: 128GB
- python 3.12.8

MarcelPiNacy-CVC, 2025-02-07:
Could you hare the entire log of the build and launch?

ihshin227, 2025-02-08:
Here is my `~/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt` file

https://github.com/ihshin227/carla_build_log/blob/main/Log.txt

ihshin227, 2025-02-10:
I solve this error.

Until now I just work on remote-ssh terminal. 

So, I try on my unbuntu pc locally and carla works without error.

Thank you Mercel, and I close this issue.

closed 2025-02-10 by ihshin227