carla-simulator/carla #8657: problem of make PythonAPI

aylady02, 2025-02-07:
Hello, after I git clone carla, an error occurred when I run make pythonapi, as follows:
~/carla$ make PythonAPI 
MakeFile build is currently disabled.

ubuntu 20.04
carla latest
Thanks in advance !

himanshupradeep, 2025-02-09:
Yes, even I am facing the same issue.

MarcelPiNacy-CVC, 2025-02-09:
Hi, the build steps for UE5 have changed. Check the new documentation here: https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/

closed 2025-02-09 by MarcelPiNacy-CVC

himanshupradeep, 2025-02-09:
Could you please guide the same way for installation in windows as well ? 
I got the error but I am doing my installation from source in windows. 

Thanks in advance

aylady02, 2025-02-10:
> Could you please guide the same way for installation in windows as well ? I got the error but I am doing my installation from source in windows.
> 
> Thanks in advance

https://www.youtube.com/watch?v=wdeHxZKWaIc
This is the video I referred to when I installed it on Windows before, I hope it can help you.

kaushik180, 2025-02-12:
> Could you please guide the same way for installation in windows as well ? I got the error but I am doing my installation from source in windows.
> 
> Thanks in advance

@himanshupradeep, did you figure the issue out? I get the same issue while trying to install Carla.

himanshupradeep, 2025-02-12:
Hello !

Older solution doesn´t work anymore. CARLA changed the way to builds in version 0.10.0.
Please follow the instructions given in this new page.

https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/

Before starting with the steps mentioned in the above page link,

1. Uninstall Visual Studio 19 or 22 or both.
2. Old Python application
3. Old CMake application
4. Delete complete CARLA folder and Unreal Engine Folder

**git clone -b ue5-dev https://github.com/carla-simulator/carla.git CarlaUE5**
(The above command will clone the new CARLAUE5 to your local machine)

BEFORE YOU START THIS, kindly add the new environmental variable with this directory attached

Environment Variable - **CARLA_UNREAL_ENGINE_PATH**
Directory - **C:\CarlaUE5\Unreal\CarlaUnreal**

**cd CarlaUE5**  (Run in CMD)
**CarlaSetup.bat** 

Run these setup script and this will take care of installing all necessary dependencies. Later, follow the instructions on the page.
My build failed at certain stage but I am working on a solution. I will update once I have completed the installation. 

Regards
Himanshu