carla-simulator/carla #8660: “carla/Memory.h”: No such file or directory

a1162750126, 2025-02-08:
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "F:\carla-0.9.15\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
E:\Programs\Python\Python38\lib\site-packages\setuptools\_distutils\cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
"e:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include -IE:\Programs\Python\Python38\include -IE:\Programs\Python\Python38\Include "-Ie:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-Ie:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-Ie:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\VS\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\um" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\shared" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\winrt" "-IC:\Program Files (x86)\Windows Kits\10\\include\10.0.22621.0\\cppwinrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-Ie:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-Ie:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22621.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-cpython-38\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
source/libcarla/libcarla.cpp(7): fatal error C1083: 无法打开包括文件: “carla/Memory.h”: No such file or directory
error: command 'e:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "F:\carla-0.9.15\PythonAPI\carla\dist"!

Jonah-1, 2025-02-11:
Rename version.h.i from \carla\LibCarla\source\carla, remove the.in, and run make pythonAPI again

w823117380, 2025-02-11:
Have you solved it

seknao, 2025-02-12:
I have the same issue. Did you find a solution?

CXLI1124, 2025-02-23:
Don't worry about it, just run make launch. After my testing, it can be used normally. In subsequent use, simply run the version corresponding to pip install carla to run it

A5Tr0tu, 2025-03-15:
> Rename version.h.i from \carla\LibCarla\source\carla, remove the.in, and run make pythonAPI again

thank you bro, you are right!!!

closed 2026-11-12 by Blyron