carla-simulator/carla #8661: Building ue5 error

Millarx, 2025-02-08:
1. ubuntu22.04
2. carla 0.10.0 at path `/home/ubuntu/projects/carla/carla-0.10.0`
3. ue：UnrealEngine-ue5-dev-carla.zip unzip to path:`/home/ubuntu/projects/carla/UnrealEngine5_carla/`
4. python: conda base python 3.12 

ContentClone.log：`fatal: destination path 'Carla' already exists and is not an empty directory.`
cmd error：
```
ubuntu@ubuntu:~/projects/carla/carla-0.10.0$ ./CarlaSetup.sh ：

Total execution time: 1.99 seconds
Unhandled exception: NullReferenceException: Object reference not set to an instance of an object.
   at EpicGames.Core.FileSystemReference.CombineStrings(DirectoryReference baseDirectory, String[] fragments) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/Shared/EpicGames.Core/FileSystemReference.cs:line 46
   at EpicGames.Core.DirectoryReference.Combine(DirectoryReference baseDirectory, String[] fragments) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/Shared/EpicGames.Core/DirectoryReference.cs:line 158
   at UnrealBuildTool.UBAExecutor.ExecuteActionsAsync(IEnumerable`1 inputActions, ILogger logger, IActionArtifactCache actionArtifactCache) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/UnrealBuildTool/Executors/UnrealBuildAccelerator/UBAExecutor.cs:line 342
   at UnrealBuildTool.ActionGraph.ExecuteActionsAsync(BuildConfiguration BuildConfiguration, List`1 ActionsToExecute, List`1 TargetDescriptors, ILogger Logger, IActionArtifactCache actionArtifactCache) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/UnrealBuildTool/System/ActionGraph.cs:line 464
   at UnrealBuildTool.BuildMode.BuildAsync(TargetMakefile[] Makefiles, List`1 TargetDescriptors, BuildConfiguration BuildConfiguration, BuildOptions Options, FileReference WriteOutdatedActionsFile, ILogger Logger, String ActionTypeFilter) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/UnrealBuildTool/Modes/BuildMode.cs:line 823
   at UnrealBuildTool.BuildMode.BuildAsync(List`1 TargetDescriptors, BuildConfiguration BuildConfiguration, ISourceFileWorkingSet WorkingSet, BuildOptions Options, FileReference WriteOutdatedActionsFile, ILogger Logger, Boolean bSkipPreBuildTargets, String ActionTypeFilter) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/UnrealBuildTool/Modes/BuildMode.cs:line 415
   at UnrealBuildTool.BuildMode.ExecuteAsync(CommandLineArguments Arguments, ILogger Logger) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/UnrealBuildTool/Modes/BuildMode.cs:line 255
   at UnrealBuildTool.UnrealBuildTool.Main(String[] ArgumentsArray) in /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Source/Programs/UnrealBuildTool/UnrealBuildTool.cs:line 687
FAILED: Unreal/CMakeFiles/carla-unreal /home/ubuntu/projects/carla/carla-0.10.0/Build/Unreal/CMakeFiles/carla-unreal 
cd /home/ubuntu/projects/carla/carla-0.10.0/Build/Unreal && /home/ubuntu/projects/carla/UnrealEngine5_carla/Engine/Build/BatchFiles/Linux/Build.sh CarlaUnreal Linux Development -project=/home/ubuntu/projects/carla/carla-0.10.0/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw
ninja: build stopped: subcommand failed.
```

MarcelPiNacy-CVC, 2025-02-20:
Hi, if you have downloaded UE5 manually don't run CarlaSetup.sh. I'd suggest you use the manual build instructions from the CARLA UE5 docs.

wlemkens, 2025-03-11:
I get the same error when running the manual build instructions. (Step `cmake --build Build --target launch` specifically)