carla-simulator/carla #8663: [BUG] Incorrect static actor bounding box extents in 0.9.15

SimonGer, 2025-02-09:
### Description
Static actors have incorrect bounding box extents when their yaw is not 0. This has been observed both on static cars (which are static.prop.mesh) and static objects like static.prop.trafficwarning. 
The issue seems to arise from the rotation values, the bounding box has an additional rotation value, which is not 0, but set to the same value as the actor rotation. It seems that the extent vector is being rotated by the rotation values, static cars with a yaw of 90° for example have swapped x and y extents and cars at different angles have extent values that deviate from the expected extents, which should be independent of rotation and around (2.5,  1, z).

### Visualization
| Static cars with 90° global yaw with flipped x and y extents | Static cars at odd angle with wrong extent values |
| - | - |
| [image] | [image] |

### Reproduce
For simplicity, these two routes can be used to reproduce the issue for static cars and the traffic warning respectively:
- [Static cars](https://github.com/autonomousvision/carla_garage/blob/leaderboard_2/data/50x36_Town13/VehicleOpensDoorTwoWays/46_0.xml)
- [Traffic warning](https://github.com/autonomousvision/carla_garage/blob/leaderboard_2/data/50x36_Town13/ConstructionObstacle/1198_3.xml)

It is also possible to reproduce this directly on the validation routes. A construction obstacle as well as some static cars at 90 degrees can be observed at the start of route 0, you can comment out the first 8 waypoints to get to some static cars which are at an angle.

The traffic warning in route 0 will yield the following results:

`static.prop.trafficwarning -70.62481689453125 -70.62481689453125 Vector3D(x=1.736055, y=0.610234, z=1.807624)`

The correct extent for a traffic warning is: `[1.186714768409729, 1.4352929592132568, 1.784761667251587]`

You can use this simple actor which prints the relevant values:
```python
import carla
from leaderboard.autoagents.autonomous_agent import AutonomousAgent
from srunner.scenariomanager.carla_data_provider import CarlaDataProvider

def get_entry_point():
  return "TestAgent"

class TestAgent(AutonomousAgent):
    def run_step(self, input_data, timestamp):

        vehicle = CarlaDataProvider.get_hero_actor()
        world = vehicle.get_world()

        static_list = world.get_actors().filter('*static*')
        for static in static_list:

            if static.type_id == "static.prop.mesh":
                name = static.attributes['mesh_path']
            elif static.type_id == "static.prop.trafficwarning":
                name = static.type_id
            else:
                continue

            print(name, static.get_transform().rotation.yaw, static.bounding_box.rotation.yaw, static.bounding_box.extent)

        control = carla.VehicleControl()
        control.steer = 0.0
        control.throttle = 0.0
        control.brake = 0.0
        control.hand_brake = False

        return control
```

Then start a simulator and run the following command:
```
python leaderboard/leaderboard/leaderboard_evaluator.py --routes={ROUTE_FILE} --agent=path_to/test_agent.py
```

Additionally, you can use a client to confirm the issue from outside the agent:
```python
import carla
client = carla.Client("localhost", 2000)
client.set_timeout(10)
world = client.get_world()

print("Meshes:")
print([(x.bounding_box.extent.x, x.bounding_box.extent.y) for x in world.get_actors().filter('*static*') if x.type_id == "static.prop.mesh"])
print("Traffic warnings:")
print([(x.bounding_box.extent.x, x.bounding_box.extent.y) for x in world.get_actors().filter('*static*') if x.type_id == "static.prop.trafficwarning"])
```

If you have any questions I am happy to help :)

PatrickPromitzer, 2025-02-10:
Hi,
my guess is that you don't use the location of the bounding box center to calculate the right center point.
```
bounding_box.location.x
bounding_box.location.y
bounding_box.location.z
```

I had similar problems and calculated the bounding box information myself at some point. It helped me with understanding the problem I had.
https://github.com/carla-simulator/carla/issues/7007

SimonGer, 2025-02-10:
Hi, 
I don't quite understand how the bounding box center should be considered for the extent values, can you elaborate on that? The issue for me is just that the extent values themselves are incorrect, so for example if a car usually has [2.5, 1, 1] it might have something like [1.4, 1.6, 1] if it is rotated in the global coordinate system. Aren't the extent values always in the actor coordinate frame and shouldn't change?

My workaround right now is to spawn all objects once with a script and storing their extents and using them as a lookup table.

PatrickPromitzer, 2025-02-10:
You need the center to draw the bounding box, with will use the center as start point.
btw. the size of the object is 2 times the extent value
[image]

The rotation of the bounding box is 0.0, but you still have to consider the actor rotation.

The actor is in the world coordinate system.
The bounding box is in the actor coordinate system.
If you want to shift the bounding box, you have to do that in the actor coordinate system.

It is the same workflow as converting LIDAR data in the global coordinate system, while it is mounted on a vehicle.
[image]
If the vehicle rotation changes, you have translation changes in the LIDAR (or in your example, bounding box)

SimonGer, 2025-02-10:
Thanks for your input, but the issue isn't directly related to drawing/visualizing the bounding boxes but only to the values of the extent vector. They shouldn't be affected by any locations or rotations I believe?

> The rotation of the bounding box is 0.0, but you still have to consider the actor rotation.

This is what I'm also referring to in the issue. The bounding box rotation is not 0 and I believe that is part of the problem. In the agent file you can see I'm printing both `static.get_transform().rotation.yaw` and `static.bounding_box.rotation.yaw`, which both yield the exact same (non-zero) value.

PatrickPromitzer, 2025-02-10:
> This is what I'm also referring to in the issue. The bounding box rotation is not 0 and I believe that is part of the problem. In the agent file you can see I'm printing both [...]

Even if you move and rotate the actor/object, the values in static.bounding_box should not change.
I made a list of each spawn-able object, and checked the static.bounding_box.rotation value of each object, and all of them had 0.0 for all 3 values.

Are you sure you look at the right values and it is not a python floating point error?

Other question.
How are you sure the values are wrong?

PatrickPromitzer, 2025-02-10:
> Are you spawning the objects manually? I think the problem might have something to do with how objects are created by the scenarios.

I used the python API to spawn the vehicle for testing, but use the ScenarioRunner for the my simulation run. 

> You can have a look at the agent I provided, I don't think there's anything wrong. The deviations can be around 1, so I don't expect this to be a floating point issue. Especially considering the example I provided in the original issue, where I am able to recreate the incorrect values by rotating the correct vector.

I can't tell if something is wrong, because I don't see output data.

> Firstly by visual inspection, as you can see in the two images, in the first the cars are far too wide compared to the cars on the road and in the second image the extents are flipped, since the cars are at 90° in the global coordinate system. Secondly, I manually spawned the objects and looked at their extent. Cars usually have an xy-extent around (2.5, 1), so values like (1.4, 1.6) or (1, 2.5) can be called incorrect I think.

The problem is, I can't tell if the visual is right. 
vehicle.ford.mustang 
x: 4.717525005340576
y: 1.894826889038086
z: 1.300939917564392

With .filter('*static*') you don't use vehicles.
```
[
static.prop.warningaccident, static.prop.doghouse, static.prop.travelcase, static.prop.trashcan05, 
static.prop.trashcan02, static.prop.trashcan01, static.prop.trashbag, static.prop.calibrator, 
static.prop.trafficwarning, static.prop.trafficcone01, static.prop.table, static.prop.streetfountain, 
static.prop.streetsign04, static.prop.streetsign, static.prop.shoppingtrolley, static.prop.shoppingbag, 
static.prop.shoppingcart, static.prop.purse, static.prop.plastictable, static.prop.plantpot07, 
static.prop.plantpot05, static.prop.plantpot04, static.prop.plantpot03, static.prop.mobile, 
static.prop.kiosk_01, static.prop.ironplank, static.prop.advertisement, static.prop.gnome, 
static.prop.gardenlamp, static.prop.dirtdebris01, static.prop.haybale, static.prop.creasedbox01, 
static.prop.container, static.prop.colacan, static.prop.motorhelmet, static.prop.clothesline, 
static.prop.foodcart, static.prop.slide, static.prop.plasticbag, static.prop.clothcontainer, 
static.prop.dirtdebris02, static.prop.chainbarrier, static.prop.creasedbox03, static.prop.vendingmachine, 
static.prop.brokentile04, static.prop.brokentile03, static.prop.brokentile02, static.prop.brokentile01, 
static.prop.plantpot02, static.prop.briefcase, static.prop.box01, static.prop.bike helmet, 
static.prop.bench03, static.prop.bench02, static.prop.trashcan04, static.prop.platformgarbage01, 
static.prop.bench01, static.prop.busstoplb, static.prop.fountain, static.prop.constructioncone, 
static.prop.bin, static.prop.mailbox, static.prop.garbage03, static.prop.wateringcan, 
static.prop.pergola, static.prop.trashcan03, static.prop.streetsign01, static.prop.garbage05, 
static.prop.garbage01, static.prop.busstop, static.prop.plantpot06, static.prop.garbage02, 
static.prop.barrel, static.prop.trafficcone02, static.prop.warningconstruction, static.prop.dirtdebris03, 
static.prop.garbage06, static.prop.swing, static.prop.box03, static.prop.plantpot01, 
static.prop.glasscontainer, static.prop.creasedbox02, static.prop.chainbarrierend, static.prop.trampoline, 
static.prop.guitarcase, static.prop.atm, static.prop.swingcouch, static.prop.box02, 
static.prop.maptable, static.prop.garbage04, static.prop.streetbarrier, static.prop.plantpot08, 
static.prop.plasticchair, static.prop.barbeque, static.prop.haybalelb, 
]

```

SimonGer, 2025-02-10:
You need a scenario where there is parked cars, which are actors of type static.prop.mesh