carla-simulator/carla #8664: Carla "MakeFile build is currently disabled."

himanshupradeep, 2025-02-09:
Hello team
I am facing this issue while executing make Python API Command in x64 Native Tools Command Prompt. 

C:\carla>make PythonAPI 
"MakeFile build is currently disabled."

I am very close to finishing the installation but with this error, I am stuck. Kindly help.

Windows 11
Lastest Carla
Build from Source on windows

PatrickPromitzer, 2025-02-10:
Hi,

Carla 0.10.0 is not using "make" commands.
https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/

```
cmake --build Build
cmake --build Build --target carla-python-api-install
cmake --build Build --target launch
```

himanshupradeep, 2025-02-10:
Hi Patrick,

Could you share the solution for windows version please ? This unfortunately didn´t work.

MarcelPiNacy-CVC, 2025-02-10:
Patrick is right, you need to read the build documentation for UE5. There is a separate page for windows there too.

himanshupradeep, 2025-02-12:
Hello !

Older solution doesn´t work anymore. CARLA changed the way to builds in version 0.10.0.
Please follow the instructions given in this new page.

https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/

Before starting with the steps mentioned in the above page link,

1. Uninstall Visual Studio 19 or 22 or both.
2. Old Python application
3. Old CMake application
4. Delete complete CARLA folder and Unreal Engine Folder

**git clone -b ue5-dev https://github.com/carla-simulator/carla.git CarlaUE5**
(The above command will clone the new CARLAUE5 to your local machine)

BEFORE YOU START THIS, kindly add the new environmental variable with this directory attached

Environment Variable - **CARLA_UNREAL_ENGINE_PATH**
Directory - **C:\CarlaUE5\Unreal\CarlaUnreal**

**cd CarlaUE5**  (Run in CMD)
**CarlaSetup.bat** 

Run these setup script and this will take care of installing all necessary dependencies. Later, follow the instructions on the page.
My build failed at certain stage but I am working on a solution. I will update once I have completed the installation. 

Regards
Himanshu

ukrio, 2025-02-21:
> Hello team I am facing this issue while executing make Python API Command in x64 Native Tools Command Prompt.
>
> C:\carla>make PythonAPI "MakeFile build is currently disabled."
>
> I am very close to [...]

same i also facing this issue can somebody help

PatrickPromitzer, 2025-02-21:
> same i also facing this issue can somebody help

Please read and try the solution to this issue before writing that something doesn't work.