carla-simulator/carla #8668: lidar label

Vish19-code, 2025-02-11:
lidar_position = carla.Transform(carla.Location(x=-46.17, y=-73.55, z=3.4), carla.Rotation(pitch=0, yaw=0))

i am placing my lidar in this location and want label data 3d object detection so it is possible and how i try but they not giving right data in the label file of the object.

lidar_bp = blueprint_library.find('sensor.lidar.ray_cast')
lidar_bp.set_attribute('range', '50')  # Adjust range as needed for coverage
lidar_bp.set_attribute('rotation_frequency', '10')
lidar_bp.set_attribute('channels', '32')

PatrickPromitzer, 2025-02-18:
Hi,
The "sensor.lidar.ray_cast" doesn't have an label.
https://carla.readthedocs.io/en/0.9.15/ref_sensors/#lidar-sensor
Array of 32-bits floats (XYZI of each point).

You are looking for "sensor.lidar.ray_cast_semantic"
https://carla.readthedocs.io/en/0.9.15/ref_sensors/#semantic-lidar-sensor

- Array containing the point cloud with instance and semantic information. For each point, four 32-bits floats are stored. XYZ coordinates.  
- cosine of the incident angle.  
- Unsigned int containing the index of the object hit.   
- Unsigned int containing the semantic tag of the object it.

PatrickPromitzer, 2025-02-18:
Which Carla version are you using?
The tags changed at version 0.9.14 

[image]

Your script looks right, but you should test it against the whole blueprint list to be sure.

Vish19-code, 2025-02-18:
CARLA Version: 0.9.15

i am using this version but same problem cyclic is not able to label 

what is the problem ?

PatrickPromitzer, 2025-02-19:
With the "sensor.lidar.ray_cast_semantic" sensor, you always get an label back, even it it is tag 0.
The sensor give you numbers and not strings.

What do you get back?
How does your code look like?
How do you convert the sensor data?

PatrickPromitzer, 2025-02-19:
This code doesn't use LIDAR data.

I can't tell the functionality of the function transform_to_sensor_coordinates(obj, lidar), but the only part you are using is the location of the sensor.
"vehicle.motorcycle", "vehicle.bicycle", and "vehicle.*two_wheel*" are not an blueprints that exists.
You say the LIDAR labels are wrong, but you are using the blueprint Ids.

Before we continue, write down what you want to do, because it doesn't look like your code fit to the question you are asking.
Check the data in different steps, because you filter the actors without knowing the actor values.

Vish19-code, 2025-02-19:
hello,

i am using infrastructure based 1 lidar and 2 cameras in and i have to label participants and make 3D label in label as well as calibration file like car, pedestrial and cyclic.

In that i have those problem that cyclic is not able to label they recognised as car and label it in label file is car not cyclic

[image]

PatrickPromitzer, 2025-02-19:
Because, vehicle.* is everything with 2 and 4 wheels, and the first if statement doesn't work.

https://carla.readthedocs.io/en/0.9.15/python_api/#carlaactor
you need actor.semantic_tags

Vish19-code, 2025-02-19:
hello, 

Thank you it works

i have another problem that i spawn the pedestrian but it going in the road like random location not in side road like regular traffic participant so what should i do?

PatrickPromitzer, 2025-02-19:
You should open a new ticket or ask in discord.
If you ask in an issue which an different topic, nobody will find the question to help you.

Blyron, 2025-08-28:
> i have another problem that i spawn the pedestrian but it going in the road like random location not in side road like regular traffic participant so what should i do?

You can teleport it manually. I think currently is the only way. Open other ticket for that purpouse as the issue of the ticket has been solved