carla-simulator/carla #8673: Vehicle Dynamics Data in CARLA

alfredgu001324, 2025-02-12:
Hi there, 

Thanks for the great simulator. I wonder how the angular velocity reading is obtained from carla, by directly reading from `CarlaDataProvider.get_hero_actor().get_angular_velocity()`, the value seems a bit weird. It can go up to 20 rad/s, but from the visualization we can see the the car is nor spinning. If I try to approximate it using the difference in heading across timesteps, the results look much better and reasonable, although the reading is still large sometimes. Also the acceleration values can go up to 1G (which is reasonable in reality racing, but not sure in CARLA ..)

The above observations are made in a scenario where the vehicle is driving at high speed (>30 m/s) but without background traffic. I just want to understand how real the physics of CARLA is and how much we should trust the vehicle dynamics data in CARLA. 

Thank you so much for everyone who may know some insights!

alfredgu001324, 2025-02-18:
Oh I see, thanks for the clarification. Do you mind explaining what is 'does not call the simulator'?

PatrickPromitzer, 2025-02-19:
The function does not make an API call to Carla.

If you send a tick, update vehicle information, spawn an actor, .... you send an API request to Carla to tell it what to do.
Some function need to make an API call and some don't, because the information where updated automatically.