carla-simulator/carla #8674: CARLA INSTALLATION --> "Engine Folder not found even after a successful build"  --->    C:\CarlaUE5\Unreal\CarlaUnreal\Engine\Build\B...Build.bat

himanshupradeep, 2025-02-13:
**C:\CarlaUE5>cmake --build Build --target launch**
[0/2] Re-checking globbed directories...
[0/5] C:\Windows\system32\cmd.exe /C "cd /D C:\CarlaUE5\Build\Unreal && C:\CarlaUE5\Unreal\CarlaUnreal\Engine\Build\B...Build.bat CarlaUnreal Win64 Development -project=C:/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscwThe system cannot find the path specified.

FAILED: Unreal/CMakeFiles/carla-unreal C:/CarlaUE5/Build/Unreal/CMakeFiles/carla-unreal
C:\Windows\system32\cmd.exe /C "cd /D C:\CarlaUE5\Build\Unreal && C:\CarlaUE5\Unreal\CarlaUnreal\Engine\Build\BatchFiles\Build.bat CarlaUnreal Win64 Development -project=C:/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw"
ninja: build stopped: subcommand failed.

Build until this step ran without any error.

**ISSUE : Unable to locate Build.Bat file.** 
I don't see Engine folder being created.

Environment Variable Setup

Variable Name : CARLA_UNREAL_ENGINE_PATH
Variable Path : C:\CarlaUE5\Unreal\CarlaUnreal

What am I doing wrong here? Kindly help.

Source Page : https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/

Regards
Himanshu Pradeep

MarcelPiNacy-CVC, 2025-02-17:
Hi,
Did you run CarlaSetup.bat, or have you manually downloaded the engine earlier?

himanshupradeep, 2025-02-18:
Hello Marcel,
To answer your question, Yes. I did run CarlaSetup.bat. I finished the installation without any errors i.e successful build. When I try to launch, I get the error as shown below.

I had earlier tried to install CARLA using this YouTube link https://youtu.be/wdeHxZKWaIc?si=rDTL1EyKzDqjwkcA where the creator has shown step by step installation guide for Unreal Engine 4 and CARLA 0.9.12 . My build failed because there is a new installation path for UE5 and newer CARLA. 

New Guide for CarlaUE5 : https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/

Before I started with the new type of installation, I manually deleted older Unreal Engine File from C folder. Wasn't that enough or is there any specific way to delete Unreal Engine. 

C:\CarlaUE5>cmake --build Build --target launch

ERROR:
[0/2] Re-checking globbed directories...
[0/5] C:\Windows\system32\cmd.exe /C "cd /D C:\CarlaUE5\Build\Unreal && C:\CarlaUE5\Unreal\CarlaUnreal\Engine\Build\B...Build.bat CarlaUnreal Win64 Development -project=C:/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscwThe system cannot find the path specified.

FAILED: Unreal/CMakeFiles/carla-unreal C:/CarlaUE5/Build/Unreal/CMakeFiles/carla-unreal
C:\Windows\system32\cmd.exe /C "cd /D C:\CarlaUE5\Build\Unreal && C:\CarlaUE5\Unreal\CarlaUnreal\Engine\Build\BatchFiles\Build.bat CarlaUnreal Win64 Development -project=C:/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw"
ninja: build stopped: subcommand failed.

Build until this step ran without any error.

ISSUE : Unable to locate Build.Bat file.
I don't see Engine folder being created.

Kindly help me to install this successfully, it would mean a lot for my student projects.

Regards
Himanshu Pradeep

MarcelPiNacy-CVC, 2025-02-18:
Have you checked whether these paths that appear in the terminal actually exist? Check your environment variables too, just in case.

himanshupradeep, 2025-02-18:
No, they don’t exist at all. Anything after Engine folder is not present in
my Unreal Engine folder. I completed deleted all applications I previously
installed and ran the setup.bat file again since it’ll install all the
necessary applications with the correct versions on its own. My build was
complete without any errors or warnings. But  I don’t know how this path is
not present at all.
Any help is appreciated.

Thanks in advance.

Regards
HIMANSHU

MarcelPiNacy-CVC, 2025-02-18:
Weird, I'd suggest you manually build UE5 and CARLA then.
First clone UE5 and run `Setup.bat && GenerateProjectFiles.bat`, then open the generated .sln with Visual Studio 2022 and build all.
On the CARLA side, use the appropriate CMake commands to build it (do not run CarlaSetup.bat, only CMake commands).

MarcelPiNacy-CVC, 2025-02-18:
To configure and build CARLA I'd suggest you use:
```
cmake --preset Development
cmake --build Build/Development
cmake --build Build/Development --target carla-python-api-install
cmake --build Build/Development --target launch
```

MarcelPiNacy-CVC, 2025-02-20:
Hi, did you have any luck using these?

MaharshiPathakCode, 2025-02-20:
Hey, I have facing a similar issue while running the prompt :

I have installed UE5.5.3 manually from the EPIC games launcher & CARLASetup.bat was perfectly installed.
I have been installing Carla in a separate volume and my VS 2022 is in C.

The above cmd is

[image]

 done already via Doc: https://carla-ue5.readthedocs.io/en/latest/build_windows_ue5/
Input> "cmake --build Build --target launch"

Output>[1/3] Compile [x64] RayCastSemanticLidar.cpp
M:\CarlaProject\CarlaUE5\Unreal\CarlaUnreal\Plugins\Carla\Source\Carla\Sensor\RayCastSemanticLidar.cpp(262): error C2039: 'ParallelLineTraceSingleByChannel': is not a member of 'UWorld'
M:\UE_5.5\Engine\Source\Runtime\Engine\Classes\Engine\World.h(855): note: see declaration of 'UWorld'
Trace file written to C:/Users/mahar/AppData/Local/UnrealBuildTool/Log.uba with size 3.2kb
Total time in Unreal Build Accelerator local executor: 16.74 seconds
Total execution time: 17.54 seconds

FAILED: Unreal/CMakeFiles/carla-unreal M:/CarlaProject/CarlaUE5/Build/Unreal/CMakeFiles/carla-unreal
C:\WINDOWS\system32\cmd.exe /C "cd /D M:\CarlaProject\CarlaUE5\Build\Unreal && M:\UE_5.5\Engine\Build\BatchFiles\Build.bat CarlaUnreal Win64 Development -project=M:/CarlaProject/CarlaUE5/Unreal/CarlaUnreal/CarlaUnreal.uproject -game -buildscw"
ninja: build stopped: subcommand failed.

MarcelPiNacy-CVC, 2025-02-20:
You must use our fork. The UE5 from Epic Games does not have some modifications we added.

MaharshiPathakCode, 2025-02-21:
ok cool, The last time I ran Carlasetup.bat, it worked perfectly and the UE was downloaded with this setup. 
the new installation I do have a set environment variable of GIT, Cmake & Make.
>UE path will added after carlasetup.bat right?

MaharshiPathakCode, 2025-02-22:
I have resolved the issue:

Solution: 1. Clean installation by uninstalling CMake, Make and Visual Studio 2022.
               2. Look at path C:\Users\mahar\AppData\Local, search for the folder "UnrealEngine, Unreal, Carla" and delete all.
               3. Set up the Environment again and also the CARLA UE path in the System Variable.
               4. Run clone cmd.
               5. After that run 
               6. CarlaSetup.bat will automatically download all the required programs and support.
               7. Note CMake will give you trouble while running Build CARLA cmd

closed 2025-06-13 by Blyron