carla-simulator/carla #8675: Build Carla 09.15 in Ubuntu 20.04

digilab2025, 2025-02-13:
I am trying to build Carla 09.15 but I am facing follwoing issue. Does anyone know how to solve it? 

`(myenv) adminknetend2@adminknetend2-System-Product-Name:~/Dokumente/carlasim/carla-0.9.15$ make PythonAPI
Util/BuildTools/Linux.mk:163: Warnung: Die Befehle für das Ziel „downloadplugins“ werden überschrieben
Util/BuildTools/Linux.mk:142: Warnung: Alte Befehle für das Ziel „downloadplugins“ werden ignoriert
Setup.sh: Retrieving boost.
--2025-02-13 11:03:44--  https://boostorg.jfrog.io/artifactory/main/release/1.80.0/source/boost_1_80_0.tar.gz
Auflösen des Hostnamens boostorg.jfrog.io (boostorg.jfrog.io) … 3.95.117.170, 18.232.172.199, 18.214.194.113
Verbindungsaufbau zu boostorg.jfrog.io (boostorg.jfrog.io)|3.95.117.170|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Moved Temporarily
Platz: https://landing.jfrog.com/reactivate-server/boostorg [folgend]
--2025-02-13 11:03:45--  https://landing.jfrog.com/reactivate-server/boostorg
Auflösen des Hostnamens landing.jfrog.com (landing.jfrog.com) … 18.232.172.199, 18.214.194.113, 3.95.117.170
Verbindungsaufbau zu landing.jfrog.com (landing.jfrog.com)|18.232.172.199|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 11534 (11K) [text/html]
Wird in »boost_1_80_0.tar.gz.3« gespeichert.

boost_1_80_0.tar.gz 100%[===================>]  11,26K  --.-KB/s    in 0s      

2025-02-13 11:03:46 (107 MB/s) - »boost_1_80_0.tar.gz.3« gespeichert [11534/11534]

Setup.sh: Extracting boost for Python 3.

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make: *** [Util/BuildTools/Linux.mk:142: setup] Fehler 2
`

PatrickPromitzer, 2025-02-18:
Hi,
the link for boost is broken

open the following file

path/to/carla/Util/BuildTools/Setup.sh

Go to line 94 and change this code

old:
```
wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
```

new:
```
wget "https://archives.boost.io/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
```

CHILLed144, 2025-05-23:
> Hi, the link for boost is broken
>
> open the following file
>
> path/to/carla/Util/BuildTools/Setup.sh
>
> Go to line 94 and change this code
>
> old:
>
> ```
> wget [...]

had same problem while building CARLA 0.9.13, this solution works. thanks