carla-simulator/carla #8686: Compilation Error: "Can't link to the standard math library" with Eigen on Ubuntu 24.04

onlywl9598, 2025-02-17:
Operating System: Ubuntu 24.04.2 LTS
Kernel Version: 6.11.0-061100-generic
Conda Environment: 
`Package             Version     Editable project location
------------------- ----------- --------------------------------------------------------
build               1.2.2.post1
carla               0.10.0      /home/only/workspace/wang/carla/CarlaUE5/PythonAPI/carla
certifi             2024.8.30
charset-normalizer  3.4.1
exceptiongroup      1.2.2
idna                3.10
importlib_metadata  8.5.0
importlib_resources 6.4.5
numpy               1.24.4
packaging           24.2
pathspec            0.12.1
pip                 25.0.1
psutil              6.1.1
pygame              2.6.1
pyproject_hooks     1.2.0
requests            2.32.3
scikit_build_core   0.10.7
setuptools          75.1.0
tomli               2.2.1
typing_extensions   4.12.2
urllib3             2.2.3
wheel               0.44.0
zipp                3.20.2`
Carla Version: Carla 0.10.0
GPU Driver: NVIDIA GeForce RTX 4060 Ti  cuda12.6  Driver Version: 560.35.05 
Compiler Version: gcc version 14.2.0 
cmake version: 3.28.3

`Can't link to the standard math library. Please report to the Eigen developers, telling them about your platform.`

compile script:
`#! /bin/bash

set -e

interactive=0
skip_prerequisites=0
launch=0
python_root=/home/xxx/workspace/anaconda3/envs/carla/bin/

workspace_path="$(dirname $(realpath "${BASH_SOURCE[-1]}"))"
echo "workspace_path=$workspace_path"

options=$(\
    getopt \
    -o "i,p,l,pyroot:" \
    --long "interactive,skip-prerequisites,launch,python-root:" \
    -n 'CarlaSetup.sh' -- "$@")

eval set -- "$options"
while true; do
    case "$1" in
        -i|--interactive)
            interactive=1
            shift
            ;;
        -p|--skip-prerequisites)
            skip_prerequisites=1
            shift
            ;;
        -l|--launch)
            launch=1
            shift
            ;;
        -pyroot|--python-root)
[... 252 lines cut ...]
-- Performing Test BOOST_FILESYSTEM_HAS_POSIX_AT_APIS
-- Performing Test BOOST_FILESYSTEM_HAS_POSIX_AT_APIS - Failed
-- Boost.Math: standalone mode OFF
-- Boost.Multiprecision: standalone mode OFF
-- Found Python: /home/only/workspace/anaconda3/envs/carla/include/python3.8 (found version "3.8.20") found components: Development NumPy Interpreter Development.Module Development.Embed 
-- Boost.Python: using Python 3.8.20 with NumPy at /home/only/workspace/anaconda3/envs/carla/lib/python3.8/site-packages/numpy/core/include
-- Found the following ICU libraries:
--   data (required): /usr/lib/x86_64-linux-gnu/libicudata.so
--   i18n (required): /usr/lib/x86_64-linux-gnu/libicui18n.so
--   uc (required): /usr/lib/x86_64-linux-gnu/libicuuc.so
-- Found ICU: /usr/include (found version "74.2") 
-- Boost.Thread: threading API is pthread
-- Performing Test EIGEN_COMPILER_SUPPORT_CPP11
-- Performing Test EIGEN_COMPILER_SUPPORT_CPP11 - Success
-- Performing Test COMPILER_SUPPORT_std=cpp03
-- Performing Test COMPILER_SUPPORT_std=cpp03 - Success
-- Performing Test standard_math_library_linked_to_automatically
-- Performing Test standard_math_library_linked_to_automatically - Failed
-- Performing Test standard_math_library_linked_to_as_m
-- Performing Test standard_math_library_linked_to_as_m - Failed
CMake Error at Build/_deps/eigen-src/CMakeLists.txt:108 (message):
  Can't link to the standard math library.  Please report to the Eigen
  developers, telling them about your platform.

-- Configuring incomplete, errors occurred!`

onlywl9598, 2025-02-18:
> Same error when trying to run on Ubuntu 24.04
I tried creating a build folder in the eigen source code under the Carla's Build/_deps directory, and running cmake .. inside the build folder to compile, and it passed completely without any errors. So, I suspect the issue lies in the build command for eigen in Carla, but I haven't been able to find exactly where the eigen compilation is triggered.

MarcelPiNacy-CVC, 2025-02-20:
Hi,
This looks like a UE5 build issue rather than an incorrect CMake command. Try building UE5 manually first.

MarcelPiNacy-CVC, 2025-02-20:
Btw please share logs using markdown or as separate files. Otherwise formatting breaks and you could also potentially mass ping random users, which has happened a couple of times.

TechCraft0, 2025-02-21:
> Hi, This looks like a UE5 build issue rather than an incorrect CMake command. Try building UE5 manually first.

Regarding the UE5 compilation issue you mentioned, I plan to attempt a manual compilation again tonight according to your guidance and will pay attention to any situations that may arise.
At the same time, I will make corresponding adjustments to the log format to ensure compliance with the recommendation of using markdown or separate files. This will help in clearly presenting the information and avoiding formatting issues. Thank you again for your help!

TechCraft0, 2025-02-23:
> Hi, This looks like a UE5 build issue rather than an incorrect CMake command. Try building UE5 manually first.

I manually executed the UE5 build, but the error still occurs when I try again.

TechCraft0, 2025-02-28:
> I'm also using Ubuntu 24.04 and I arrived at the same issue. First attempt simply running CarlaSetup.sh got to the "make" stage of the build where it tries to compile UnrealEngine_carla. It fails to compile around these Modules:
> 
> ```
> [156/4247] Compile Module.SignalProcessing.1.cpp
> [202/4247] Compile Module.Chaos.10.cpp
> [203/4247] Compile Module.Chaos.1.cpp
> [207/4247] Compile Module.Chaos.12.cpp
> [210/4247] Compile Module.Chaos.14.cpp
> [211/4247] Compile Module.Chaos.13.cpp
> [212/4247] Compile Module.Chaos.16.cpp
> [213/4247] Compile Module.Chaos.17.cpp
> [214/4247] Compile Module.Chaos.15.cpp
> [217/4247] Compile Module.Chaos.11.cpp
> [222/4247] Compile Module.Chaos.5.cpp
> [225/4247] Compile Module.Chaos.7.cpp
> [451/4247] Compile Module.Engine.27.cpp
> [455/4247] Compile Module.Engine.26.cpp
> [456/4247] Compile Module.Engine.29.cpp
> [459/4247] Compile Module.Engine.33.cpp
> [465/4247] Compile Module.Engine.42.cpp
> [493/4247] Compile Module.Engine.65.cpp
> [636/4247] Compile Module.GeometryCollectionEngine.cpp
> [775/4247] Compile Module.Niagara.10.cpp
> [896/4247] Compile Module.GeometryCache.cpp
> [1107/4247] Compile Module.ChaosCloth.cpp
> [1141/4247] Compile Module.ChaosClothAssetEngine.cpp
> [1339/4247] Compile Module.NNERuntimeBasicCpu.cpp
> [1437/4247] Compile Module.AnimNextAnimGraph.cpp
> [1609/4247] Compile Module.Learning.cpp
> ```
> 
> Until it crashed around [2895/4247].
> 
> After a while I tried building UnrealEngine_carla separately and got the same error again (surprise) but this time I read somewhere that that I should just rerun the .../UnrealEngine_carla: "make" and magically it seemed to have resolved itself.
> 
> Of course I have no idea if this is related to the issue - however when trying to run .../CarlaUE5: CarlaSetup.sh afterwards it fails at:
> 
> ```
> -- Performing Test standard_math_library_linked_to_as_m - Failed
> CMake Error at Build/_deps/eigen-src/CMakeLists.txt:108 (message):
>   Can't link to the standard math library.  Please report to the Eigen
>   developers, telling them about your platform.
> 
> 
> -- Configuring incomplete, errors occurred
> ```
> 
> And trying to manually run:
> 
> ```
> cmake -G Ninja -S . -B Build --toolchain=$PWD/CMake/LinuxToolchain.cmake \
> -DLAUNCH_ARGS="-prefernvidia" -DCMAKE_BUILD_TYPE=Release -DENABLE_ROS2=ON
> ```
> 
> Lands me in the same error.
> 
> I'd much appreciate some help with this.

I'm not sure how to resolve this either. I manually compiled UE5.5 alone, and everything worked fine. Then, I went into the Carla directory to compile, but the same error occurred again.

crblaur, 2025-03-03:
I'm encountering the same issue. Have you found any solution?

TechCraft0, 2025-03-04:
> I'm encountering the same issue. Have you found any solution?

I currently do not have a solution.

crblaur, 2025-03-04:
After spending 1 week trying to run the carla simulator, I realized the problem might have been the os. So after downgraded to Ubuntu 22.04 I could have successfully run carla @TechCraft0

TechCraft0, 2025-03-04:
> After spending 1 week trying to run the carla simulator, I realized the problem might have been the os. So after downgraded to Ubuntu 22.04 I could have successfully run carla [@TechCraft0](https://github.com/TechCraft0)

The main issue is that after downgrading the system, many things need to be reinstalled.

tanasis-d, 2025-09-17:
> Got the same error even with **Ubuntu 22.04.** But in that case the Unreal Engine has been forked directly from Epic Games. Issue is resolved by using Carla's own fork of Unreal Engine (https://github.com/CarlaUnreal/UnrealEngine)

I am having the same issue in Ubuntu 22.04 and I am using the Carla's fork of UE5. Anyone figured this out?