carla-simulator/carla #8691: Carla "Makefile" is currently disabled

NiccoloSalvi, 2025-02-19:
Hi everyone,
I'm trying to install Carla 0.9.14 on Ubuntu 20.04 OS.
I successfully downloaded the 0.9.14 assets, but when I run `make PythonAPI` I encounter the problem that Makefile is currently disabled. That's the content of the Makefile: 
```
launch PythonAPI LibCarla CarlaUnrealEditor launch-only: FORCE
	@echo "MakeFile build is currently disabled."
FORCE: ;

# include Util/BuildTools/Vars.mk
# ifeq ($(OS),Windows_NT)
# include Util/BuildTools/Windows.mk
# else
# include Util/BuildTools/Linux.mk
# endif
```

How can I fix it? Please note that in `path/to/carla/Util/BuildTools` there is no file named `Linux.mk`

Thanks

PatrickPromitzer, 2025-02-20:
Hi,
If you want to install Carla version 0.9.14, you have to switch to the right git branch.
The current Carla version is 0.10.0

MarcelPiNacy-CVC, 2025-02-20:
Patrick is right, I'm closing this issue.

closed 2025-02-20 by MarcelPiNacy-CVC