carla-simulator/carla #8695: make PythonAPI Error on Windows10

kyoa, 2025-02-20:
`J:\UnrealProjects\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "J:\UnrealProjects\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "J:\UnrealProjects\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "J:\UnrealProjects\carla\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "J:\UnrealProjects\carla\Build\zlib-source\".
CMake Error: The source directory "J:/UnrealProjects/carla/Build/zlib-source"" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1`
I'm using Windows, I have installed the newest version python, Cmake and make following the [instruction](https://carla.readthedocs.io/en/latest/build_windows/)

MarcelPiNacy-CVC, 2025-02-20:
Hi, have you tried deleting the build folder?

kyoa, 2025-02-21:
> Hi, have you tried deleting the build folder?

Yes, but get same error

kyoa, 2025-02-21:
> Hi, have you tried deleting the build folder?

I have changed CMake version to 3.31.5, problem resolved.

closed 2025-02-21 by kyoa