carla-simulator/carla #8696: Generate not properly - Digital Twins CARLA

IrfanAli91170, 2025-02-20:
> @JarvisLee0423 Hey, I encountered the same issue. I resolved it by following the instructions [here](https://github.com/carla-simulator/carla/issues/6913#issuecomment-2380016146). I hope that helps.
> 
> 
> 
> @rohanNkhaire Thanks for your kind reply! I successfully used the instructions in the link you shared to generate the map from the example OSM, like the images below:
> 
> [image]
> 
> However, I still face a problem, which is that, after generating the map, CARLA will give me two map assets, one named Colorado and another named Colorado_Tile_0_0, like the image illustrated below:
> 
> [image]
> 
> It seems that all the geometric assets will be saved in Colorado_Tile_0_0 map, and all the world parameters assets will be saved in Colorado map, like the world light setting and playable setting. In this case, if I load the Colorado_Tile_0_0 map, the windows will be all dark until I manually put a direct light component inside the map. This situation is shown in the images below:
> 
> Directly load Colorado_Tile_0_0 map:
> 
> [image]
> 
> After I put a direct light component manually:
> 
> [image]
> 
> Meanwhile, the error will occur if I push the play button, cause the playable component is not in the Colorado_Tile_0_0 map.

> 
> After that, if I load the Colorado map, the window will only show the default sky and does not contain any buildings or roads, like the image demonstrated below:
> 
> [image]
> 
> 
> 
> Therefore, I would like to know if you have ever faced this problem before. If yes, could you tell me how to fix it? Or are there any methods to merge these two maps? It seems a little bit complex and stupid if I need to manually add all the world parameter settings (light, playable, etc.) for all maps generated by OSMWorldRenered. 

 _Originally posted by @JarvisLee0423 in [#8283](https://github.com/carla-simulator/carla/issues/8283#issuecomment-2434471667)_

I have also the same issue here. Please, can someone help me ?

[image]

[image]

Blyron, 2025-02-20:
This is expected behaviour, you have to press play in base level which is empty. Largemap maanger will populate the level properly

IrfanAli91170, 2025-02-21:
If it's a map divided in tiles, can I include these or we have to include only the base level ?

I have this line in the "DefaultGame.ini" : +MapsToCook=(FilePath="/Game/CustomMaps/SatoryTest")

This is the path to the folder SatoryTest, which contains SatoryTest.umap and others tiles maps

But when I launch CARLA binary build, there are only default maps.

Maybe, i have done mistake to only specify the folder and not .umap file (which is /Game/CustomMaps/SatoryTest/SatoryTest) and maybe should I include also the tiles .umap also ?

Blyron, 2025-02-21:
I have found a bug. Please, set your map as default map. When your base level is included the sublevels are automatically included, as you have to play the original one and not sub levelss-

To set your default map, edit DefaultEngine.ini and set there your map

IrfanAli91170, 2025-02-21:
I have a crash when i tried 'make package' with my custom map :  

[image]

I followed your instructions, but I don't know why.

Blyron, 2025-02-21:
It seems to be a DGT tool bug

IrfanAli91170, 2025-02-21:
OK, because I need to make package to share with my colleagues for my work and it's very urgent. 

Thanks for helping me

Blyron, 2025-02-21:
You can try to fix it on your own. Or make your colleagues build the Sim and share umap and assets. We are not fixing it very soon.

IrfanAli91170, 2025-02-21:
https://github.com/carla-simulator/carla/issues/8265#issue-2577884907

IrfanAli91170, 2025-02-24:
LogCook: Display: Inisetting is different for Linux Engine /Script/EngineSettings.GameMapsSettings GameDefaultMap 0, value /Game/CustomMaps/SatoryTest/SatoryTest.SatoryTest != /Game/CustomMaps/SatoryTest/SatoryTest invalidating cook

But when i change to "/Game/CustomMaps/SatoryTest/SatoryTest" :
LogCook: Display: Inisetting is different for Linux Engine /Script/EngineSettings.GameMapsSettings GameDefaultMap 0, value /Game/CustomMaps/SatoryTest/SatoryTest!= /Game/CustomMaps/SatoryTest/SatoryTest.SatoryTest invalidating cook

I don't understand ....

Again, i have the crash of 'make package'.

This is from Diagnostics.txt :

Generating report for minidump

Application version 4.26.2.0
 ... built from changelist 0

OS version Linux 6.8.0-53-generic (network name: carla)
Running 4 x86_64 processors (4 logical cores)
Exception was "SIGSEGV: invalid attempt to read memory at address 0x0000000000000070"

<SOURCE START>
<SOURCE END>

<CALLSTACK START>
libUE4Editor-Engine.so!FDistanceFieldVolumeTexture::Release() [/home/scarlet/UE4.26/Engine/Source/Runtime/Engine/Private/DistanceFieldAtlas.cpp:841]
libUE4Editor-Engine.so!FDistanceFieldAsyncQueue::ProcessAsyncTasks() [/home/scarlet/UE4.26/Engine/Source/Runtime/Engine/Private/DistanceFieldAtlas.cpp:1279]
libUE4Editor-Engine.so!FDistanceFieldAsyncQueue::BlockUntilBuildComplete(UStaticMesh*, bool) [/home/scarlet/UE4.26/Engine/Source/Runtime/Engine/Private/DistanceFieldAtlas.cpp:1155]
libUE4Editor-Engine.so!FStaticMeshRenderData::Serialize(FArchive&, UStaticMesh*, bool) [/home/scarlet/UE4.26/Engine/Source/Runtime/Engine/Private/StaticMesh.cpp:1521]
libUE4Editor-Engine.so!UStaticMesh::Serialize(FArchive&) [/home/scarlet/UE4.26/Engine/Source/Runtime/Engine/Private/StaticMesh.cpp:4697]
libUE4Editor-CoreUObject.so!FArchiveSaveTagExports::ProcessTaggedObjects() [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/SavePackage.cpp:343]
libUE4Editor-CoreUObject.so!FPackageExportTagger::TagPackageExports(FArchiveSaveTagExports&, bool) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/SavePackage.cpp:1785]
libUE4Editor-CoreUObject.so!UPackage::Save(UPackage*, UObject*, EObjectFlags, char16_t const*, FOutputDevice*, FLinkerNull*, bool, bool, unsigned int, ITargetPlatform const*, FDateTime const&, bool, FAr$
libUE4Editor-UnrealEd.so!UEditorEngine::Save(UPackage*, UObject*, EObjectFlags, char16_t const*, FOutputDevice*, FLinkerNull*, bool, bool, unsigned int, ITargetPlatform const*, FDateTime const&, bool, F$
libUE4Editor-UnrealEd.so!UCookOnTheFlyServer::SaveCookedPackage(UE::Cook::FPackageData&, unsigned int, TArrayView<ITargetPlatform const* const, int> const&, TArray<FSavePackageResultStruct, TSizedDefaul$
libUE4Editor-UnrealEd.so!UCookOnTheFlyServer::PumpSaves(UE::Cook::FTickStackData&, unsigned int) [/home/scarlet/UE4.26/Engine/Source/Editor/UnrealEd/Private/CookOnTheFlyServer.cpp:2347]
libUE4Editor-UnrealEd.so!UCookOnTheFlyServer::TickCookOnTheSide(float, unsigned int&, ECookTickFlags) [/home/scarlet/UE4.26/Engine/Source/Editor/UnrealEd/Private/CookOnTheFlyServer.cpp:1152]
libUE4Editor-UnrealEd.so!UCookCommandlet::CookByTheBook(TArray<ITargetPlatform*, TSizedDefaultAllocator<32> > const&) [/home/scarlet/UE4.26/Engine/Source/Editor/UnrealEd/Private/Commandlets/CookCommandl$
libUE4Editor-UnrealEd.so!UCookCommandlet::Main(FString const&) [/home/scarlet/UE4.26/Engine/Source/Editor/UnrealEd/Private/Commandlets/CookCommandlet.cpp:622]
UE4Editor!FEngineLoop::PreInitPostStartupScreen(char16_t const*) [/home/scarlet/UE4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3369]
UE4Editor!GuardedMain(char16_t const*) [/home/scarlet/UE4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/scarlet/UE4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xe6)
UE4Editor!_start()

<CALLSTACK END>

0 loaded modules

Report end!

This is from

IrfanAli91170, 2025-02-24:
OK, no problem ! Thank you for your response

Blyron, 2025-02-24:
You can have a look at this, https://github.com/carla-simulator/carla/pull/8620,  to have docker image with built ue4 editor and you can share the map with the imported assets

Blyron, 2025-02-25:
Right now is not possible for you. As it is bug within CARLA, which hasn't been solved yet

IrfanAli91170, 2025-02-25:
I don't know if it is due to memory leaks, i have tried to generate another map (really really small) and there is the same issue.

I know, i want to search the issue behind, myself and i need to understand what is this about.

Thanks in advance

Blyron, 2025-02-25:
The issue must be related with the generated geometry on the roads. Something is broken there. Why? I do not know yet

IrfanAli91170, 2025-02-25:
I think it is the same concept for Town11 and Town12, which are large maps (divided by tiles). I will try to include these maps in the packaged build to watch if it works or not.

Blyron, 2025-02-25:
Largemaps concept is okay and it works perfectly. The issue is related to the GENERATED GEOMETRY, AKA, ROADS and SIDEWALKS.

IrfanAli91170, 2025-02-25:
The folder generated by the Digital Twin Tool is called 'CustomMaps', and it is located outside the 'Carla' folder. Both 'CustomMaps' and 'Carla' are at the same level inside the 'Content' directory.

Here is the hierarchy:

Content/  
│── Carla/       (Default CARLA assets and maps)  
│── CustomMaps/  (Maps generated by the Digital Twin Tool)  

This means that the 'CustomMaps' folder is not inside 'Carla', but rather placed alongside it within the 'Content' directory.

Note : It has been done automatically by the Digital Twin tool.

Maybe do you think it will affects ?

Blyron-AeroSim, 2025-02-25:
> The folder generated by the Digital Twin Tool is called 'CustomMaps', and it is located outside the 'Carla' folder. Both 'CustomMaps' and 'Carla' are at the same level inside the 'Content' [...]

Again.  The issue is related to the **procedurally generated geometry, AKA,  road and sidelwaks.**

IrfanAli91170, 2025-02-25:
If you say that the problem comes from the road geometry, it could be related to the XODR files.

If I try to find an area with a straight road, do you think it will work?

IrfanAli91170, 2025-02-26:
https://github.com/carla-simulator/carla/issues/8265

Blyron, 2025-02-26:
It should work.

IrfanAli91170, 2025-02-27:
Hello, i have tried with a straight road in Colorado for example.

It doesn't work.....

Note : the area i choose (no buildings, no roads with geometric variable shapes, only one straight road).

Thanks you in advance

IrfanAli91170, 2025-02-27:
And again, i just tried a map without roads, sidewalks, no buildings, only forest....

Same problem when cooking.

IrfanAli91170, 2025-03-03:
Hello !

Following up on our previous discussion, I'm currently using the dev branch of Digital Twin Tool, and I wanted to check if there are any updates or planned fixes regarding the make package command.

Would it be possible to get notified if any improvements or corrections are made? Or is there a recommended way to track changes related to this issue (such as a specific tag, issue, or notification system)?

Thanks in advance for your help and for all your work!

IrfanAli91170, 2025-04-10:
Hello ! 

I came here to ask if there are any updates on the issue fix....

Blyron, 2025-04-10:
No news. Sorry for this

IrfanAli91170, 2025-05-07:
Hello,

I'm following up on issue, which I opened two months ago regarding a problem exporting maps created with the Digital Twins Tool in CARLA.

I'm still facing this issue, and it's currently blocking a critical part of my project involving urban traffic simulation. I'm in urgent need of a solution or workaround to move forward.

Have you had a chance to test this on your end or identify any potential cause?
Even a partial response would be greatly appreciated so I can adjust my approach accordingly.

Thank you very much in advance, and kudos again for your excellent work on CARLA.

Blyron, 2025-05-07:
Yeah we are able to identify the cause are the generated static meshes distance fields. 
We are not able to fix it properlly.

Blyron, 2025-05-07:
We need more time

Blyron, 2025-08-25:
Hello, 
If you update update the engine fork this should be fixed.

shamuspatry-helm, 2025-12-11:
I think I'm having the same problem. The fix links to a non-existent (or non-public) repo?