carla-simulator/carla #8698: Add all .whl files to 0.10.0 Linux Release

jbeck9, 2025-02-21:
CARLA version: 0.10.0
Platform/OS: Ubuntu 22.04

Referencing https://github.com/carla-simulator/carla/issues/8544, the Ubuntu releases still do not contain .whl files for the other versions of Python that are supported in 0.10.0 (3.8, 3.9, 3.11, 3.12).

Adding those to the release, adding them to pypi, or providing instructions for building only those .whl files from source would be appreciated.

Blyron, 2025-02-21:
They should be on the package, we will be updating it.

NestSteel, 2025-03-04:
> They should be on the package, we will be updating it.
Unfortunately, as of 2025.03.04, the downloaded Ubuntu carla 0.10 still only contains carla-0.10.0-cp310-cp310-linux_x86_64.whl, I hope you will include it in the plan soon, I would appreciate it

qhaas, 2025-03-06:
Adding 0.10.0 to pypi would be great, could just pip install without having to download the full tarball:  https://pypi.org/project/carla/

NestSteel, 2025-03-07:
Great, the latest version supported by https://pypi.org/project/carla/ right now is carla 0.9.15, hopefully updated to 0.10.0 soon

Blyron, 2025-03-07:
Until we have uploaded which we will be soonish next week 
https://drive.google.com/drive/folders/1Czpn49MiVBBgFLEZ1A31Rn4hmBOrA1b4

glopezdiest, 2025-03-13:
Sorry for the delay, we've updated the release link with all the wheels.

Closing this but feel free to reopen it if there are any issues still remaining

closed 2025-03-13 by glopezdiest