carla-simulator/carla #8699: make PythonAPI Error on win10 0.9.15

kyoa, 2025-02-21:
Environment:
OS: win10
carla: 0.9.15
CarlaUnreal: 4.26.2
Python: x64 3.13.2
CMake: 3.31.5
Make: GNU Make 3.81

> Building Custom Rule J:/UnrealProjects/carla/Build/recast-src/CMakeLists.txt
  1>
  -- Install configuration: "Release"
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/version.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/pkgconfig/recastnavigation.pc
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/cmake/recastnavigation/recastnavigation-targets.cmake
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/cmake/recastnavigation/recastnavigation-targets-relea
  se.cmake
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/cmake/recastnavigation/recastnavigation-config.cmake
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/cmake/recastnavigation/recastnavigation-config-versio
  n.cmake
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/DebugUtils.lib
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DebugDraw.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourDebugDraw.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/RecastDebugDraw.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/RecastDump.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/Detour.lib
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourAlloc.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourAssert.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourCommon.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourMath.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourNavMesh.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourNavMeshBuilder.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourNavMeshQuery.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourNode.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourStatus.h                               -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/DetourCrowd.lib                                         -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourCrowd.h                                -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourLocalBoundary.h                        -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourObstacleAvoidance.h                    -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourPathCorridor.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourPathQueue.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourProximityGrid.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/DetourTileCache.lib
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourTileCache.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/DetourTileCacheBuilder.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/lib/Recast.lib
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/Recast.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/RecastAlloc.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/include/recast/RecastAssert.h
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/bin/RecastBuilder.exe
  -- Installing: J:/UnrealProjects/carla/Build/recast-install/../../Util/DockerUtils/dist/RecastBuilder.exe
此时不应有 neq。

    -[install_recast]: "Recast & Detour" has been successfully installed in "J:\UnrealProjects\carla\Build\recast-install\"
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "J:\UnrealProjects\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 16
    -[install_boost]: Retrieving boost.
    -[install_boost]: Extracting boost from "boost_1_80_0.zip", this can take a while...
New-Object : 使用“3”个参数调用“.ctor”时发生异常:“找不到中央目录结尾记录。”
所在位置 C:\Windows\system32\WindowsPowerShell\v1.0\Modules\Microsoft.PowerShell.Archive\Microsoft.PowerShell.Archive.p
sm1:934 字符: 23
+ ... ipArchive = New-Object -TypeName System.IO.Compression.ZipArchive -Ar ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [New-Object]，MethodInvocationException
    + FullyQualifiedErrorId : ConstructorInvokedThrowException,Microsoft.PowerShell.Commands.NewObjectCommand

    -[install_boost]: Removing "boost_1_80_0.zip"
系统找不到指定的文件。
系统找不到指定的路径。
    -[install_boost]: Generating build...
'bootstrap.bat' 不是内部或外部命令，也不是可运行的程序
或批处理文件。

    -[install_boost]: [BOOTSTRAP ERROR] An error ocurred while executing "bootstrap.bat".
    -[install_boost]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 错误 1

Reason1520, 2025-02-26:
I'm having the same problem compiling carla version 0.9.12 in win11

Reason1520, 2025-02-26:
I have solved this problem, you can go to the official website of boost.com https://archives.boost.io/release/1.80.0/source/, download the zip and unzip it to the Build folder, then re-enter the command make pythonapi.

ShallowMode, 2025-03-04:
@Reason1520 Hello, I'm trying to build carla 0.9.15 in win 11. 
I'm having same problem when I entered Update.bat. Can you help me, please?