carla-simulator/carla #8703: issue: could you release a new carla UE4 version with latest CarlaUE4-Linux-Shipping

zczjx, 2025-02-23:
- requirement:
could you release a new carla UE4 version with latest CarlaUE4-Linux-Shipping

- issue description:

log:

4.26.2-0+++UE4+Release-4.26 5220	复制	
Disabling core dumps.	
Signal 11 caught.	
Malloc Size=65538 LargeMemoryPool0ffset=65554	
CommonUnixCrashHandler: Signal=11	
Malloc Size=131160LargeMemoryPoo10ffset=196744	
Malloc Size=131160LargeMemoryPool0ffset=327928	
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.	
Segmentation fault (core dumped)

- fixing CL

https://github.com/CarlaUnreal/UnrealEngine/commits/carla/

the Fixing commnts should be in the UE4 repo recently, but no new binary release for  CarlaUE4-Linux-Shipping

could you help to deliver a new release about UE4 for this issue

qhaas, 2025-03-07:
It would be useful to continue to get bug fixes in the form of new UE4 branch-based binaries until the UE5 branch supports all the major features of the UE4 branch (e.g. DVS Camera #8724 ), we recently had to revert to 0.9.15 because of a need for said sensor model.

VioletMirror, 2025-03-25:
Not only Linux but also windows release would be very useful as #8547 which is supposed to deal with fixing night builds still has not been closed.

Blyron, 2025-06-26:
This should be working now and we are doing nightly builds

closed 2025-06-26 by Blyron