carla-simulator/carla #8713: Exporting a map error: `-[Package]: [ERROR] Carla Version is not set`

zhang2336, 2025-02-27:
# make_package_error

2025年02月27日 10:39:32 UTC+8:00

## 1. Errors that occur

When I used the `blender-driving-scenario-creator-v0.28.2` plugin for **Blender** to complete Carla's custom map setup and it successfully ran in UE4

https://github.com/user-attachments/assets/466b4181-77ce-4de0-9640-40124fad3bfd

Trying to export a map of `xiong_map_test` that is already working to **Package installation**, I refer to [Content authoring - maps - CARLA Simulator]( https://carla.readthedocs.io/en/latest/tuto_content_authoring_maps/#exporting-a-map)

After I executed the command, an error occurred

```cmd
make package ARGS="--packages=xiong_test_map"
```

[image]

Output of the error

```cmd
  osm2odr.vcxproj -> F:\carla-ue4-dev\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  1>
  -- Install configuration: "Release"
已复制         1 个文件。
已复制         1 个文件。
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "F:\carla-ue4-dev\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
creating build\bdist.win-amd64\egg
creating build\bdist.win-amd64\egg\carla
copying build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd -> build\bdist.win-amd64\egg\carla
copying build\lib.win-amd64-3.8\carla\__init__.py -> build\bdist.win-amd64\egg\carla
byte-compiling build\bdist.win-amd64\egg\carla\__init__.py to __init__.cpython-38.pyc
creating stub loader for carla\libcarla.cp38-win_amd64.pyd
byte-compiling build\bdist.win-amd64\egg\carla\libcarla.py to libcarla.cpython-38.pyc
creating build\bdist.win-amd64\egg\EGG-INFO
copying source\carla.egg-info\PKG-INFO -> build\bdist.win-amd64\egg\EGG-INFO
copying source\carla.egg-info\SOURCES.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying source\carla.egg-info\dependency_links.txt -> build\bdist.win-amd64\egg\EGG-INFO
copying source\carla.egg-info\top_level.txt -> build\bdist.win-amd64\egg\EGG-INFO
writing build\bdist.win-amd64\egg\EGG-INFO\native_libs.txt
zip_safe flag not set; analyzing archive contents...
carla.__pycache__.libcarla.cpython-38: module references __file__
creating 'dist\carla-0.9.15-py3.8-win-amd64.egg' and adding 'build\bdist.win-amd64\egg' to it
removing 'build\bdist.win-amd64\egg' (and everything under it)
running bdist_wheel
running build
running build_ext
installing to build\bdist.win-amd64\wheel
running install
running install_lib
creating build\bdist.win-amd64\wheel
creating build\bdist.win-amd64\wheel\carla
copying build\lib.win-amd64-3.8\carla\libcarla.cp38-win_amd64.pyd -> build\bdist.win-amd64\wheel\.\carla
copying build\lib.win-amd64-3.8\carla\__init__.py -> build\bdist.win-amd64\wheel\.\carla
running install_egg_info
Copying source\carla.egg-info to build\bdist.win-amd64\wheel\.\carla-0.9.15-py3.8.egg-info
running install_scripts
C:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\wheel\_bdist_wheel.py:124: RuntimeWarning: Config variable 'Py_DEBUG' is unset, Python ABI tag may be incorrect
  if get_flag("Py_DEBUG", hasattr(sys, "gettotalrefcount"), warn=(impl == "cp")):
creating build\bdist.win-amd64\wheel\carla-0.9.15.dist-info\WHEEL
creating 'dist\carla-0.9.15-cp38-cp38-win_amd64.whl' and adding 'build\bdist.win-amd64\wheel' to it
adding 'carla/__init__.py'
adding 'carla/libcarla.cp38-win_amd64.pyd'
adding 'carla-0.9.15.dist-info/METADATA'
adding 'carla-0.9.15.dist-info/WHEEL'
adding 'carla-0.9.15.dist-info/top_level.txt'
adding 'carla-0.9.15.dist-info/RECORD'
removing build\bdist.win-amd64\wheel

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "F:\carla-ue4-dev\PythonAPI\carla\dist"!
-[BuildCarlaUE4]: [Batch params]: --at-least-write-optionalmodules --package=xiong_test_map
--at-least-write-optionalmodules
--package
xiong_test_map
ECHO 处于关闭状态。
-[Package]: [Batch params]: --ue-version 4.26 --package=xiong_test_map
fatal: not a git repository (or any of the parent directories): .git

-[Package]: [ERROR] Carla Version is not set
make: *** [package] 错误 1
```

## 2. `make package` settings

I followed the settings of [Content authoring - maps - CARLA Simulator](https://carla.readthedocs.io/en/latest/tuto_content_authoring_maps/#exporting-a-map) and put the various files of the generated `xiong_test_map` under the path:

```cmd
F:\carla-ue4-dev\Unreal\CarlaUE4\Content\Carla\Maps\xiong_test_map>tree /f
卷 新加卷 的文件夹 PATH 列表
卷序列号为 1C4F-6435
F:.
│  road_asphalt.uasset
│  road_mark_white.uasset
│  xiong_test_map.umap
│
├─OpenDrive
│      xiong_test_map.xodr
│
└─TM
        xiong_test_map.bin
```

and xiong_test_map.Package to save the file:

[image]

Here `xiong_test_map. Package.json` content

```json
{
    "props": [],
    "maps": [
        {
            "name": "xiong_test_map",
            "path": "/Game/Carla/Maps/xiong_test_map",
            "use_carla_materials": false
        }
    ]
}
```

## Questions I would like to ask

1. What does the `-[Package]: [ERROR] Carla Version is not set` output by cmd mean here, I executed `make package --help` and I didn't find the content of the specified version number, how to solve it?

2. Is my `make package` set up correctly?

In addition, I am in China, I may not be able to give you feedback in time due to the problem of time zone, and the grammar of this issue may not be smooth, please understand

I'm sorry for the inconvenience, but I'm very grateful for the help you have given me anyway

zhang2336, 2025-03-17:
It's been three weeks since this issue was posted, and I'm still very desperate to find an answer to my question, and I'm hoping an engineer can answer it for me

SuHui0802, 2025-04-02:
我也有这个问题，请问你解决了嘛

Blyron, 2025-04-03:
We are dealing with other projects and new features. For the time being you can set your map as the default map instead of Town10 and do the make package.

zhang2336, 2025-04-09:
> 我也有这个问题，请问你解决了嘛

I've tested successfully, but I'm going to use the 'https' method to download the repository

```sh
git clone <https>
```

The main thing is to have a '.git' folder in the root directory of the repository, where the version information is stored

</br>

我已经测试成功了，但是要使用 `https` 的方式下载存储库

```sh
git clone <https>
```

主要就是在存储库的根目录下要有 `.git` 文件夹，版本信息是存储在这里面的