carla-simulator/carla #8718: Windows Build Doesn't Working

ShallowMode, 2025-02-28:
I downloaded the source code 0.9.15 and following Windows build Document.
But Update.bat function didn't working. I think the Download File URL is problem. Please help.

MarcelPiNacy-CVC, 2025-03-03:
Hi, could you share the console output? You should also pull the latest changes from ue4-dev, we recently added some fixes for an issue that sounds like yours.

ShallowMode, 2025-03-04:
[image]

ShallowMode, 2025-03-04:
@MarcelPiNacy-CVC 
C:\Users\psh\Desktop\Reinforce\carla>make PythonAPI GENERATOR="Visual Studio 17 2022"
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Users\psh\Desktop\Reinforce\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "Visual Studio 17 2022" --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 17 2022"
-[Setup]: Install directory:  "C:\Users\psh\Desktop\Reinforce\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Users\psh\Desktop\Reinforce\carla\Build\"
    -[install_zlib]: Retrieving zlib.
    -[install_zlib]: Extracting zlib from "zlib-1.2.13.zip".
    -[install_zlib]: Removing "zlib-1.2.13.zip"
    -[install_zlib]: Creating "C:\Users\psh\Desktop\Reinforce\carla\Build\zlib-source\build"
CMake Error: The source directory "C:/Users/psh/Desktop/Reinforce/carla/Build/zlib-source"" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 오류 1

ShallowMode, 2025-03-04:
@MarcelPiNacy-CVC I pulled the latest changes from ue4-dev. But, when i tried make PythonAPI the error is occured.

build\boost\bin.v2\libs\date_time\build\install\boost_date_time-config-version.cmake
        1개 파일이 복사되었습니다.
...skipped 61 targets...
...updated 16539 targets...

    -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] 오류 1

MarcelPiNacy-CVC, 2025-03-04:
Hi again, please share the entire output log, ideally in markdown format to avoid unwanted pings.

ShallowMode, 2025-03-05:
[image]

MarcelPiNacy-CVC, 2025-03-05:
Seems like a download problem or a build folder corruption issue. I'd suggest you delete the build folder for now, then run `make setup` and share the entire output in text form here (not images, just attach it as a file).

ShallowMode, 2025-03-05:
Okay. Also, cmake can't download Xerces 3.2.4.

ShallowMode, 2025-03-05:
[carla_build_log.txt](https://github.com/user-attachments/files/19088250/carla_build_log.txt)

ShallowMode, 2025-03-05:
[carla_build_log.txt](https://github.com/user-attachments/files/19088283/carla_build_log.txt)

ShallowMode, 2025-03-07:
> Seems like a download problem or a build folder corruption issue. I'd suggest you delete the build folder for now, then run `make setup` and share the entire output in text form here (not images, just attach it as a file).

Ifrahaha, 2025-04-21:
Hey did you solve the issue ?

dx6nxr, 2025-07-08:
Had the same issue with b2.exe error, any amount of clearing the build folder, downloading the boost library by myself, installing it by myself didnt help. Apparently boost cannot be build with msvc14.3 and is the only package that I had to build with previous version of VS and therefore msvc14.2. Installing ue4 carla is hot mess since the release of ue5.

Blyron, 2025-08-04:
Windows builds are working for us, in our CI/CD check your installation enviroment, check latest ue4-dev branch

closed 2026-11-12 by Blyron