carla-simulator/carla #8724: DVS Camera removed in 0.10.0 / UE5?

qhaas, 2025-03-04:
While [still documented in 0.10.0](https://github.com/carla-simulator/carla/blob/0.10.0/Docs/ref_sensors.md#dvs-camera), the [example was removed](https://github.com/carla-simulator/carla/commit/e2dc71c39aa70b1ec4e632860df3531215bc1ef9): `Removed DVS and added semantic lidar to mcwt`

A grep for `sensor.camera.dvs` in commit ada75f920642e18cace0a9f85ecf9d4077ddb531 only returns the documentation and some tests.  Was the DVS camera removed with the move to UE5?
```
$ git log -1
commit ada75f920642e18cace0a9f85ecf9d4077ddb531 (HEAD, tag: 0.10.0)
Author: Joel Moriana <joel.moriana@gmail.com>
Date:   Wed Dec 18 15:07:43 2024 +0100

    Release dockerfile updated
$ grep -lIr sensor.camera.dvs .
./Docs/bp_library.md
./Docs/ref_sensors.md
./PythonAPI/test/API/prop_control.py
./PythonAPI/test/smoke/test_sensor_tick_time.py
```

Blyron, 2025-03-05:
DVS was removed for the moment as it was broken. We will try to reimplement it on the future. We will remove the doc references to it.

glopezdiest, 2025-03-07:
Hello, yes, the DVS camera is not supported in 0.10.0. Sorry for the confusion, we missed those docs there. @MattRoweEAIF  can we fix them?

closed 2025-03-07 by glopezdiest

qhaas, 2025-03-30:
Thanks for the update.  Should this issue be re-opened to track the adding of DVS back to ue5-dev or is that not planned?

Blyron, 2025-03-30:
Not in short time

qhaas, 2025-12-09:
The fix in the now abandoned(?) pull request #8823 has stopped working with newer versions built with the newer CARLA UE5 fork.