carla-simulator/carla #8725: Building package not working as intended in 0.10.0

felsager, 2025-03-04:
When following "https://carla-ue5.readthedocs.io/en/latest/build_linux_ue5/" to package Carla UE5 the builds succeeds and "Carla-0.10.0-Linux-Shipping" can be found in the package folder. But it only has the "PythonAPI" and "Tools" directories, missing "CarlaUnreal" and "Engine" which can be found in the official downloadable zip "Carla-0.10.0-Linux-Shipping" https://github.com/carla-simulator/carla/releases. Are there any known problems regarding packaging for Linux?

MarcelPiNacy-CVC, 2025-03-11:
Hi, in what system are you running?

MarcelPiNacy-CVC, 2025-03-11:
Yeah, there are some reported issues in that version of ubuntu. I'd suggest you switch to the recommended Ubuntu version by UE5, which is 22.04. Alternatively you can wait for us to find a workaround for this problem, but I can't give you a date yet unfortunately.

felsager, 2025-03-11:
I will try building it on Ubuntu 22.04 in a Docker container. Meanwhile I will try to find a workaround for it myself. If you have any idea of what the problem could be, you are welcome to point me in that direction. If I find a good workaround I will make a pull request

felsager, 2025-03-12:
So far I have narrowed down the problem to be with the cooking process. 
When using "cmake --build Build --target package" and the process gets to executing 

"dotnet AutomationTool.dll BuildCookRun -project=/home/batman/carla_stripped/Unreal/CarlaUnreal/CarlaUnreal.uproject -nocompileeditor -nop4 -cook -stage -archive -package -iterate -clientconfig=Shipping -TargetPlatform=Linux -Platform=Linux -prereqs -build -stagingdirectory=/home/batman/carla_stripped/Build/Package/StagedBuilds/Carla-0.10.0-Linux-Shipping -archivedirectory=/home/batman/carla_stripped/Build/Package/Carla-0.10.0-Linux-Shipping
/home/batman/UnrealEngine5_carla/Engine/Build/BatchFiles/Linux/../../../Binaries/ThirdParty/DotNet/8.0.300/linux-x64/dotnet" 

[problematic_log.txt](https://github.com/user-attachments/files/19214584/problematic_log.txt)

as also shown in the log, but doesn't output anything like it is supposed to. 
Cooking manually with "/home/batman/UnrealEngine5_carla/Engine//Binaries/ThirdParty/DotNet/8.0.300/linux-x64/dotnet /home/batman/UnrealEngine5_carla/Engine/Binaries/DotNET/AutomationTool/AutomationTool.dll BuildCookRun -project=/home/batman/carla_stripped/Unreal/CarlaUnreal/CarlaUnreal.uproject -nocompileeditor -nop4 -cook -stage -archive -package -iterate -clientconfig=Shipping -TargetPlatform=Linux -Platform=Linux -prereqs -build -log -Verbose" works if I exclude the "-build" argument. If I include it I will get the following error "Setting up ProjectParams for /home/batman/carla_stripped/Unreal/CarlaUnreal/CarlaUnreal.uproject
/home/batman/carla_stripped/Unreal/CarlaUnreal/CarlaUnreal.uproject does not look like uproject file but no targets have been found!
(see /home/batman/UnrealEngine5_carla/Engine/Programs/AutomationTool/Saved/Logs/Log.txt for full exception trace)"
How does this line up with the problems you have found?