carla-simulator/carla #8728: Error when running make PythonAPI to build the source version

curtisjock, 2025-03-05:
I am trying to build Carla from source following the documentation. I have followed all of the step exactly and installed all of the prerequisites. I installed UE4 and the builds and runs. I cloned the Carla repository and ran Update.bat for the current version. I have also added the environment variable. when i try to run the make PythonAPI i get the following error:

C:\Carla\Carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\Carla\Carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\Carla\Carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\Carla\Carla\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "C:\Carla\Carla\Build\zlib-source\".
CMake Error: The source directory "C:/Carla/Carla/Build/zlib-source"" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

parvathydev1, 2025-03-21:
hi, was this issue resolved?

UniAdaRepo, 2025-03-25:
Hi, i have the same issue. The folder 'zlib-source' does exist in my folder but it gives 'not exist' error. Could someone help?

curtisjock, 2025-03-27:
This issue has not been resolved.

tkdgus0782, 2025-03-29:
hi, i suffered by that error too
but when i downgrade cmake version 4.00 to 3.xx, it resolved. please try it

parvathydev1, 2025-03-30:
Thankyou very much I will let you know after trying

closed 2025-04-10 by MarcelPiNacy-CVC