carla-simulator/carla #8729: Native ROS interfase is not found by MATLAB

Youjin1985, 2025-03-06:
Steps to reproduce:
1. launch  CarlaUnreal.exe --ros2
2. In matlab, execute 'rosinit'
3. In python script, spawn actor with ROS topic name [as said in documentation](https://carla-ue5.readthedocs.io/en/latest/ros2_native/)
4. in matlab, execute 'rostopic list'

actual result:
/rosout
/tf

expected result:
...
/carla/ego/vehicle_control_cmd

Youjin1985, 2025-03-06:
So question is, what is the address of CARLA ROS Master? It is not printed on launch with --ros2 argument

Youjin1985, 2025-03-06:
I found that running with or without --ros2 paratemer does not affect which ports are open at all, is this native interface working on Windows?

Blyron, 2025-03-07:
ROS2 is not working on Windows.

closed 2025-03-07 by Blyron